MicrosoftDocs/windows-driver-docs #1613: Specifics about OID data is lacking on this and other pages. Here, I was left wondering whether the vendor string needed to be ASCII, UTF-8, UTF-16, etc. On other OID pages, documentation fails to describe whether the data is a 32-bit value, a 64-bit value, or even an array of 32-bit values (for OID_GEN_SUPPORTED_LIST, for instance). Information can be found be searching through sample drivers, but actual documentation of these OID details would be appreciated. Sorry if such exists elsewhere and I have failed to discover it. If details do exist elsewhere, perhaps a link to such details on these pages would prove helpful.

druske, 2019-06-14:
[Enter feedback here]

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: bf31e680-3d78-7dd3-d9c7-8b9e4d606fa6
* Version Independent ID: 1de60554-fb57-ffec-21e0-b5f25c9a01da
* Content: [OID_GEN_VENDOR_DESCRIPTION - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/network/oid-gen-vendor-description)
* Content Source: [windows-driver-docs-pr/network/oid-gen-vendor-description.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/network/oid-gen-vendor-description.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @duncanmacmichael
* Microsoft Alias: **dumacmic**

druske, 2019-06-14:
Specifics about OID data is lacking on this and other pages. Here, I was left wondering whether the vendor string needed to be ASCII, UTF-8, UTF-16, etc. On other OID pages, documentation fails to describe whether the data is a 32-bit value, a 64-bit value, or even an array of 32-bit values (for OID_GEN_SUPPORTED_LIST, for instance). Information can be found be searching through sample drivers, but actual documentation of these OID details would be appreciated. Sorry if such exists elsewhere and I have failed to discover it. If details do exist elsewhere, perhaps a link to such details on these pages would prove helpful.

duncanmacmichael, 2019-06-15:
Hi @druske, thanks for the questions. 

To answer the first question about strings, the string should be a null-terminated Unicode string, technically UTF-16LE. According to Windows Internals, Windows itself stores and processes most internal text strings as Unicode characters within the kernel. For example, WDF always uses Unicode, as described on [Using String Objects](https://docs.microsoft.com/en-us/windows-hardware/drivers/wdf/using-string-objects). So, you should use Unicode here when populating the **InformationBuffer** field in the [**NDIS_OID_REQUEST**](https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/content/ndis/ns-ndis-_ndis_oid_request) structure during the call to [*MiniportOidRequest*](https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/content/ndis/nc-ndis-miniport_oid_request).

The key to your second question is in the Remarks section of [OID_GEN_SUPPORTED_LIST](https://docs.microsoft.com/en-us/windows-hardware/drivers/network/oid-gen-supported-list): "NDIS 6.0 and later miniport drivers do not receive this OID request. NDIS handles this OID with a cached value that miniport drivers supply during initialization." 

But, you're right in that the page doesn't say how to do this during initialization, so I've added a note there. In essence, this cached value is specified by the **SupportedOidList** member of the [**NDIS_MINIPORT_GENERAL_ATTRIBUTES**](https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/content/ndis/ns-ndis-_ndis_miniport_adapter_general_attributes) structure, which the driver passes to [**NdisMSetMiniportAttributes**](https://docs.microsoft.com/windows-hardware/drivers/ddi/content/ndis/nf-ndis-ndismsetminiportattributes). The **SupportedOidList** member is a PNDIS_OID type, or a pointer to an array of NDIS_OID values. As defined in the Ntddndis.h header, NDIS_OID is an alias of a ULONG because OIDs themselves are merely ULONG `#define` codes.

Thanks for helping to improve the docs. If there are other OID pages where you feel some information is lacking or needs a link, please feel free to open up an issue on each page in question.

closed 2019-06-15 by duncanmacmichael

druske, 2019-06-17:
The driver sample at https://github.com/microsoft/Windows-driver-samples/blob/master/network/ndis/mux/driver/60/miniport.c responds to OID_GEN_VENDOR_DESCRIPTION with a string comprised of 8-bit UCHARs (see the definition of VendorDesc in that file).