MicrosoftDocs/windows-driver-docs #1617: WinDbg analysis and identification

LuigiBruno, 2019-06-14:
I've just analyzed a DMP file that recorded this bugcheck: WinDbg version 10.0.10586.567 AMD64 didn't identify the bugcheck

Unknown bugcheck code (19c)
Unknown bugcheck description
Arguments:
Arg1: 0000000000000050
Arg2: ffffb58f3dd24080

WinDbg Preview version 10.0.18869.1002 AMD64 did identify it

WIN32K_POWER_WATCHDOG_TIMEOUT (19c)
Win32k did not turn the monitor on in a timely manner.
Arguments:
Arg1: 0000000000000050, Calling monitor driver to power on.
Arg2: ffffb58f3dd24080, Pointer to the power request worker thread.
Arg3: 0000000000000000
Arg4: 0000000000000000
Arg3: 0000000000000000
Arg4: 0000000000000000

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 08c5af02-f2c9-e39f-39ed-e32f2acd4706
* Version Independent ID: 07ab0139-9b05-54cf-c87e-4408e8c3fd22
* Content: [Bug Check 0x19C WIN32K_POWER_WATCHDOG_TIMEOUT - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/debugger/bug-check-0x19c--win32k-power-watchdog-timeout#feedback)
* Content Source: [windows-driver-docs-pr/debugger/bug-check-0x19c--win32k-power-watchdog-timeout.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/debugger/bug-check-0x19c--win32k-power-watchdog-timeout.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @DOMARS
* Microsoft Alias: **domars**

rodwiddowson, 2019-06-15:
Might be better over in https://github.com/microsoftfeedback/WinDbg-Feedback

LuigiBruno, 2019-06-17:
Thanks @rodwiddowson.

DOMARS, 2019-06-20:
@LuigiBruno - It would be great if you could open an issue on the new git feedback hub or follow this link to file product feedback using the Windows Feedback Hub: 
http://aka.ms/dexex 
I don't believe this issue is docs related, so closing this issue.

closed 2019-06-20 by DOMARS

LuigiBruno, 2019-06-25:
> @LuigiBruno - It would be great if you could open an issue on the new git feedback hub or follow this link to file product feedback using the Windows Feedback Hub:
> http://aka.ms/dexex
> I don't believe this issue is docs related, so closing this issue.

Thanks @DOMARS. I've done. it.