MicrosoftDocs/windows-driver-docs #1636: Retpoline enabled?

ttstanly, 2019-06-25:
Hi, 
According to:
https://techcommunity.microsoft.com/t5/Hardware-Dev-Center/Upcoming-Hardware-Dev-Center-changes-that-enable-support-for/ba-p/504574

How can we build a driver with Retpoline enable?

Thanks

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: e888e07f-1323-59a9-2c2f-14e9f814f552
* Version Independent ID: de2aa427-004a-b1a8-270d-cf13d0c9ed91
* Content: [How to limit or expand a driver’s distribution based on Windows version - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/dashboard/limit-driver-distribution#feedback)
* Content Source: [windows-driver-docs-pr/dashboard/limit-driver-distribution.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/dashboard/limit-driver-distribution.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @EliotSeattle
* Microsoft Alias: **eliotgra**

EliotSeattle, 2019-06-25:
Hi @ttstanly. Thanks for taking the time to ask about this.

For information on Retpoline see the blog post **Mitigating Spectre variant 2 with Retpoline on Windows**. It's at https://techcommunity.microsoft.com/t5/Windows-Kernel-Internals/Mitigating-Spectre-variant-2-with-Retpoline-on-Windows/ba-p/295618.

I think that contains the info you're looking for.

All the best.

Eliot

closed 2019-06-25 by EliotSeattle

ttstanly, 2019-06-26:
Hi Eliot,

I have read this blog before. However, I didn't find any information about how to build a driver with Retpoline enable or disable. I can't find any properties in the project file (vcxproj) for vs2017 solutions. Even I can't know if this is work for UMDF and KMDF both or only KMDF. 

The blog in dev-center that I mentioned:
https://techcommunity.microsoft.com/t5/Hardware-Dev-Center/Upcoming-Hardware-Dev-Center-changes-that-enable-support-for/ba-p/504574

Here is mentioned:
"Effective May 1 2019:  In order to support Retpoline compiled binaries on Hardware Dev Center and to protect the Windows Driver ecosystem, Microsoft will be treating Retpoline compiled binaries submitted to Hardware Dev Center differently than non-Retpoline compiled binaries.  Partners who wish to target an OS lower than RS5 x64 and Server 2019 should create a separate submission that includes non-Retpoline compiled binaries."

It looks like we can build the driver with or without Retpoline compiled. But I can't find any information on 'HOW TO' compile for disable/enable Retpoline driver.

Thanks for the time.

Stanly