MicrosoftDocs/windows-driver-docs #1638: how to compile driver with enable/disable Retopline

ttstanly, 2019-06-26:
#1636 
@EliotSeattle 
Hi Eliot,

I have read this blog before. However, I didn't find any information about how to build a driver with Retpoline enable or disable. I can't find any properties in the project file (vcxproj) for vs2017 solutions. Even I can't know if this is work for UMDF and KMDF both or only KMDF.

The blog in dev-center that I mentioned:
https://techcommunity.microsoft.com/t5/Hardware-Dev-Center/Upcoming-Hardware-Dev-Center-changes-that-enable-support-for/ba-p/504574

Here is mentioned:
"Effective May 1 2019: In order to support Retpoline compiled binaries on Hardware Dev Center and to protect the Windows Driver ecosystem, Microsoft will be treating Retpoline compiled binaries submitted to Hardware Dev Center differently than non-Retpoline compiled binaries. Partners who wish to target an OS lower than RS5 x64 and Server 2019 should create a separate submission that includes non-Retpoline compiled binaries."

It looks like we can build the driver with or without Retpoline compiled. But I can't find any information on 'HOW TO' compile for disable/enable Retpoline driver.

Thanks for the time.

Stanly

DOMARS, 2021-01-27:
@ttstanly - Thanks for highlighting this gap. We are working to gather more information on this issue.

DOMARS, 2021-03-18:
@ttstanly - Thanks for this git issue on the retpoline compiler optimizations. From what I understand, the information on enabling this option is being made available to those partners that would most benefit from the related performance increases that it would provide, under targeted situations. If you have a Microsoft contact, you might try requesting that information through that channel, to see if it would be appropriate for your development work. Thanks again for bringing up this issue, and we are looking at adding additional information to the blog post to better clarify the targeted use of this option.

closed 2021-03-18 by DOMARS