MicrosoftDocs/windows-driver-docs #1644: Sample has incorrect includes

FriskyDev, 2019-06-27:
Should be:
#define KDEXT_64BIT 
#include &lt;wdbgexts.h&gt; 
#include &lt;dbgeng.h&gt;

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: a082c77a-75fc-d11f-a8f8-544a5c6aa3f3
* Version Independent ID: 56414b73-068c-337c-6afd-20077c82d318
* Content: [Writing DbgEng Extension Code - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/debugger/writing-dbgeng-extension-code#feedback)
* Content Source: [windows-driver-docs-pr/debugger/writing-dbgeng-extension-code.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/debugger/writing-dbgeng-extension-code.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @DOMARS
* Microsoft Alias: **domars**

FriskyDev, 2019-06-27:
Also, I just realized that they are out of order. If you compile them in the current order, you get an error. But, if you reverse the order, then no error.

FriskyDev, 2019-06-27:
You will need to define the global yourself (see comments above about the name):
WINDBG_EXTENSION_APIS ExtensionApis;

And you will need the following libraries to link with this and the required APIs to load it:
#pragma comment(lib, "dbghelp")
#pragma comment(lib, "dbgeng")

DOMARS, 2019-10-25:
@FriskyDev - Thanks for pointing out the issue with the order of the includes, I fixed that in the document. On the linking as there are multiple ways to specify the files to link to such as in the IDE, I'm not going to add the pragama's right now.

closed 2019-10-25 by DOMARS