MicrosoftDocs/windows-driver-docs #1651: Links are broken

yves-dolce, 2019-07-03:
[Enter feedback here]

All the links on that page take me to a **404 - Page not found**.

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: c16f66f8-babb-8d23-8ce5-52f9b87c26b1
* Version Independent ID: 49578792-b8e0-8a37-515e-790378ea7dfa
* Content: [Extracting Value Data by Specifying Its Usage - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/hid/extracting-value-data-by-specifying-its-usage)
* Content Source: [windows-driver-docs-pr/hid/extracting-value-data-by-specifying-its-usage.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/hid/extracting-value-data-by-specifying-its-usage.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @duncanmacmichael
* Microsoft Alias: **dumacmic**

duncanmacmichael, 2019-07-03:
Hi @yves-dolce, sorry for the inconvenience. This is a known issue with this header at the moment; we're working on getting it back up and running. I'll circle back to this thread when we have it resolved.

duncanmacmichael, 2019-07-29:
Hi @yves-dolce, thanks for your patience. We've resolved the internal bug that was preventing the Hidpi.h header's content from rendering, so all links are now live.

closed 2019-07-29 by duncanmacmichael