MicrosoftDocs/windows-driver-docs #1656: what's patch?

shi-yan, 2019-07-09:
this page mentions patch-location lists without defining what patch is? and why patch-location lists are needed?

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 9fe7a6d3-1468-9d41-1996-0b72591ea10a
* Version Independent ID: 31321a15-b44a-357c-3225-e00c053ffb5e
* Content: [Using the Guaranteed Contract DMA Buffer Model - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/display/using-the-guaranteed-contract-dma-buffer-model#feedback)
* Content Source: [windows-driver-docs-pr/display/using-the-guaranteed-contract-dma-buffer-model.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/display/using-the-guaranteed-contract-dma-buffer-model.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @lorihollasch
* Microsoft Alias: **loriwhip**

shi-yan, 2019-07-09:
correct me if I'm wrong, the patch list holds allocations that are current resides on the video memory with a physical address?

Sorry if the definition of "patch" is actually mentioned in the document. But my overall feedback for the document is that it lacks the "big picture". And more diagrams could really help.

lorihollasch, 2019-10-11:
@shi-yan, thanks for your feedback. I added a quick definition for patch list. If you click on the pfnRenderDb link, you can learn a lot more. I've made a note about your request for more diagrams. I'm not sure that we'll be able to prioritize diagram work for this feature given that it is for Vista, but we will keep this in mind for future work.

closed 2019-10-11 by lorihollasch