MicrosoftDocs/windows-driver-docs #1657: The latest symbols for ntoskrnl.exe (ntkrnlmp.pdb) aren't very good anymore.

stephenscalzo, 2019-07-09:
Hi.

The debugging symbols for ntoskrnl.exe (ntkrnlmp.pdb) used to be much better in 2018 but throughout 2019 it has been awkward. The debugging symbols are no longer as complete as they used to be.

For example, even KiSystemStartup isn't named in the debugging symbols anymore. I let IDA reach out to acquire the symbols from the Microsoft Symbol Server like I always have except now I end up with several thousand APIs which are unnamed whereas older debugging symbols would have provided information about them.

Why was there a sudden redacting of information and what can be done about this?

Kind Regards,
Stephen.

stephenscalzo, 2019-07-09:
I've attached two images to this post.

The first image (**oldsymbols.png**) shows many APIs named with the debugging symbols which were available in 2018.

The second image (**newsymbols.png**) shows many APIs unnamed with the debugging symbols for **Windows 10 Enterprise (1809)**.

Note, I manually re-named some of the kernel APIs which were unnamed with the debugging symbols in newsymbols.png: *KiServiceTablesLocked*, *BcpDisplayEarlyBugCheckScreen*, *BgpFwInitializeLock*, *BgpConsoleGetFontName* and *BcpFindMessage*.  I had to check the disassembly and information such as the segment, start address, length, locals and arguments to know which ones I could re-name correctly.

[image]
[image]

It is possible that I have made a mistake however I let IDA reach out to the Microsoft Symbols Server and acquire them for me like I always have. Offline debugging symbols are neither provided anymore so this is convenient.

Furthermore, Windows 10 Home (1903) is being difficult for debugging ntoskrnl.exe because I keep getting mismatched symbols when trying to reverse-engineer. There might be an issue with the symbols for the version I have on the Microsoft Symbols Server, food for thought.

Let me know, any help is much appreciated.

Kind Regards,
Stephen.

DOMARS, 2019-07-09:
@stephenscalzo - - Thanks for reaching out to the Windows Driver Developer Documentation team. For product input on the public symbols follow this link to file product feedback using the Windows Feedback Hub: 
http://aka.ms/dexex 
The Feedback Hub allows customers to file and up vote product issues. The link automatically sets the product category to Debugging Tools for Windows.
Thanks again for reaching out to the Windows Driver Developer Documentation team.

closed 2019-07-09 by DOMARS

rodwiddowson, 2019-07-09:
Or perhaps windbg-feedback (but that is may not help for the usual symbol nightmare)

stephenscalzo, 2019-07-09:
The issue has been resolved.

The issue wasn't a fault of Microsoft but of my own because the Debug Interface Access (DIA) SDK wasn't installed. I wasn't explicitly aware of this requirement because it must have been present on my environments in the past by installation of other software - for example, Visual Studio may have automatically installed it for me in the past.

The reason the information from the old debugging symbols was available with an older copy of ntoskrnl.exe was because I was using the IDA database file which had everything packed inside of it. 

If anyone else has been experiencing the same issue with IDA, make sure that you've got the DIA SDK installed.

Kind Regards,
Stephen.

rodwiddowson, 2019-07-09:
@stephenscalzo Thanks for the FU.  its appreciated