MicrosoftDocs/windows-driver-docs #1660: 503 Service Unavailable on committing the package to partner center

tanmayee16, 2019-07-11:
I am using the hardware API to automate the process of submitting package for signing drivers. 
However so far I got success only once after committing. The python code is as follows: 
ingestionConnection = http.client.HTTPSConnection("manage.devcenter.microsoft.com")
ingestionConnection.request("POST","/v1.0/my/hardware/products/{0}/submissions/{1}/commit".format(inAppProductId,subProductId),"", headers=headers)
commitIapResponse = ingestionConnection.getresponse()
print(commitIapResponse.status)
print(commitIapResponse.reason)

The output is 
503
Service Unavailable

Please help resolve the issue asap!

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 58d4fba4-34f5-f000-fd4a-1acf9de277d0
* Version Independent ID: 638d0b07-faaa-1529-a6d5-3cd7a6b87838
* Content: [Commit a product submission - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/dashboard/commit-a-product-submission#feedback)
* Content Source: [windows-driver-docs-pr/dashboard/commit-a-product-submission.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/dashboard/commit-a-product-submission.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @balapv
* Microsoft Alias: **balapv**

tanmayee16, 2019-07-11:
As the commit is unsuccessful, the product id appears on the the Driver List Page (all) with the Submission Status =' NotSet'. If clicked on the product id the following page is displayed: 
[image]

EliotSeattle, 2019-07-11:
Hi @tanmayee16 

The Hardware Dev Center Support folks can help investigate.

1. **First**, sign into your hardware dev center account, and then 
2. Open a Hardware Dev Center support case for investigation. 
3. Go to 
[https://docs.microsoft.com/windows-hardware/drivers/dashboard/hardware-dashboard-faq](https://docs.microsoft.com/windows-hardware/drivers/dashboard/hardware-dashboard-faq)
and look for: 

**How do I contact Partner Center Support?**
If you are having problems accessing the Dashboard or need Dashboard Support please open a support ticket here: [https://developer.microsoft.com/windows/support](https://developer.microsoft.com/windows/support). 

Select **Contact us**, **Dashboard issue**, and then **Hardware submissions & signing (all OS version)** from the dropdown menu.

Thanks,

Eliot

closed 2019-07-11 by EliotSeattle