MicrosoftDocs/windows-driver-docs #1666: Ok nobody cares about windows 8 how about getting setup to do USB on windows 10 1809

studleylee, 2019-07-16:
Ok nobody cares about windows 8 how about getting setup to do USB on windows 10 1809

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 6c644605-a088-2559-f11f-5afa168e1e20
* Version Independent ID: e19385cd-b7cf-58a9-6276-0ebd4596e185
* Content: [Index of Windows Driver Kit Tools - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/devtest/index-of-windows-driver-kit-tools#what-s-changed-in-the-wdk-for-windows8)
* Content Source: [windows-driver-docs-pr/devtest/index-of-windows-driver-kit-tools.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/devtest/index-of-windows-driver-kit-tools.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @EliotSeattle
* Microsoft Alias: **eliotgra**

EliotSeattle, 2019-07-16:
Hi @studleylee 

Thanks for pointing out that this page needs some updating. I will add an item to my backlog.

If I understand your comment, you might find [WinUSB (Winusb.sys) Installation](https://docs.microsoft.com/windows-hardware/drivers/usbcon/winusb-installation) helpful. Did you have another question about the documentation on this page? 

Thanks,

Eliot

studleylee, 2019-07-16:
Reply sorry for my gruffness. I've been trying to create the mentioned
WinUsb template project using WDK for 1809 and the latest Visual Studio to
no avail.
I'll check that link you mention. Thank You!

EliotSeattle, 2019-07-17:
Closing barring further questions. Thanks so much, @studleylee

closed 2019-07-17 by EliotSeattle