MicrosoftDocs/windows-driver-docs #1668: test issue

tedhudek, 2019-07-18:
This is a test issue.

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 2c5fa76b-6c2e-8f5e-7e96-ceb1c726842b
* Version Independent ID: 48a29c52-9907-9b1a-1dc2-3365c2d021d0
* Content: [Getting Started with Universal Windows drivers - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/develop/getting-started-with-universal-drivers)
* Content Source: [windows-driver-docs-pr/develop/getting-started-with-universal-drivers.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/develop/getting-started-with-universal-drivers.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @tedhudek
* Microsoft Alias: **tedhudek**

bharney0, 2019-08-01:
#pri1

bharney0, 2019-08-05:
#please-close

bharney0, 2019-08-05:
#pri2

closed 2019-08-05 by EliotSeattle