MicrosoftDocs/windows-driver-docs #1674: Links to Digilent site and JJG technologies need to be updated

saiprasad77, 2019-07-19:
Hi,

I need to order some UART/SPI adapter boards but the link to JJG technologies listed on this page isn't working. Could you please share the working link to order them?

I also see Digilient phasing out (EOL) the Nexys 2 board. Is the Nexys 7 board compatible with UART/SPI adapter board?

Thanks in advance, Sai

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 38f16fae-1b9d-6f21-dbd5-e5fa22118685
* Version Independent ID: 4e09a919-7b21-31a8-7369-78b40cea209d
* Content: [Buy hardware for using MITT - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/spb/multi-interface-test-tool--mitt--#feedback)
* Content Source: [windows-driver-docs-pr/spb/multi-interface-test-tool--mitt--.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/spb/multi-interface-test-tool--mitt--.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @EliotSeattle
* Microsoft Alias: **eliotgra**

EliotSeattle, 2019-07-19:
Hi @saiprasad77 

Thanks for alerting me to this. I fixed the problem that kept the links from functioning. Unfortunately, I'm not getting any response from the JjG Technologies website. I'll have to look into that further.

grahamm78, 2020-04-29:
The links as written just need .htm added to them to work

http://www.jjgtechnologies.com/UART-SPI -> http://www.jjgtechnologies.com/UART-SPI.htm
http://www.jjgtechnologies.com/GPIO  -> http://www.jjgtechnologies.com/GPIO.htm
http://www.jjgtechnologies.com/I2C -> http://www.jjgtechnologies.com/I2C.htm
http://www.jjgtechnologies.com/mcatt -> http://www.jjgtechnologies.com/mcatt.htm

EliotSeattle, 2020-04-29:
Fixed in the source.
I need to go wash the egg from my face.
Thanks for pointing this out, @grahamm78 
Eliot

closed 2020-04-29 by EliotSeattle