MicrosoftDocs/windows-driver-docs #1677: Provisioning failure

JasonOldWoo, 2019-07-22:
[Enter feedback here]
An error occured while deploying files to the target machine for test "Configuring computer settings (possible reboot)":

could not find part of path “C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\VC\Redist\MSVC\14.21.27702\debug_nonredist\X64\Microsoft.VC141.DebugCRT”

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: e51ceaee-8e97-0a2a-1657-43850e68abf2
* Version Independent ID: d061c118-bdb1-f7ad-8aec-ccee9b9c3f7c
* Content: [Troubleshooting Driver Deployment, Testing and Debugging - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/develop/troubleshooting-configuration-of-driver-deployment--testing-and-debugging)
* Content Source: [windows-driver-docs-pr/develop/troubleshooting-configuration-of-driver-deployment--testing-and-debugging.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/develop/troubleshooting-configuration-of-driver-deployment--testing-and-debugging.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @tedhudek
* Microsoft Alias: **tedhudek**

tedhudek, 2019-10-15:
Hi @jasonoldwoo, apologies for the delay.  Can you provide any more info specific to the doc page?  Is there a change you are recommending or incorrect guidance?

tedhudek, 2019-10-16:
Closing this one out for now.  Please reopen with additional details if needed.

closed 2019-10-16 by tedhudek