MicrosoftDocs/windows-driver-docs #1678: C:\>kdnet <HostComputerIPAddress> <YourDebugPort> this command should change to C:\>kdnet <target ip address> <YourDebugPort>

prabbhu, 2019-07-22:
[Enter feedback here]

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 5e51b628-bb73-f32f-5a81-fefce12a4b49
* Version Independent ID: e07e2607-fd13-5517-510c-77bf717125c8
* Content: [Setting Up KDNET Network Kernel Debugging Automatically - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/debugger/setting-up-a-network-debugging-connection-automatically#feedback)
* Content Source: [windows-driver-docs-pr/debugger/setting-up-a-network-debugging-connection-automatically.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/debugger/setting-up-a-network-debugging-connection-automatically.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @DOMARS
* Microsoft Alias: **domars**

c0nd3v, 2019-08-20:
Pretty sure you're wrong lol

DOMARS, 2019-08-23:
In the step described you need to provide the host ip address, so this step is correct. Thanks for the input, and good luck in your work with the debugger and Windows.

closed 2019-08-23 by DOMARS