MicrosoftDocs/windows-driver-docs #1687: WDM Driver Fails SDV When Linking to Microsoft Framework(s)

dwkaiser, 2019-07-26:
Quoting from the documentation page:

"If you have a WDM driver or library and want to run SDV, the driver or library must not link to one of the following class framework libraries."

In my current case the WDM driver must link to smclib.lib.

(1) What prevents SDV from running in this case?
(2) Is there currently a "work around" process available to enable SDV running in this case?
(3) If not, how will the SDV related requirements for the Windows HLK be met in this case?
(4) Is there a process for the community to be able to assist in modifications to SDV and/or the framework libraries so that SDV will run successfully in this case?

Thank you.
---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: b5fd6a7f-6aab-f9b0-e4ab-ea54383adc80
* Version Independent ID: b3bd9ec2-cc7e-98c6-b445-045f2ac4a481
* Content: [Determining if Static Driver Verifier supports your driver or library - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/devtest/determining-if-static-driver-verifier-supports-your-driver-or-library)
* Content Source: [windows-driver-docs-pr/devtest/determining-if-static-driver-verifier-supports-your-driver-or-library.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/devtest/determining-if-static-driver-verifier-supports-your-driver-or-library.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @EliotSeattle
* Microsoft Alias: **eliotgra**

dbc60, 2019-08-01:
To whom it may concern, it would be nice to have some clarity. I have a minifilter, so it must link to fltmgr.lib. Do I have any options for static analysis? If not, are there any other static or dynamic test tools I could apply?

FWIW, I do put the minifilter through the HCK test studio (I have to support it running on Server 2012 R2 and beyond). I'm pretty sure the HCK and HLK test suites don't provide the same level of tests the SDV does.

EliotSeattle, 2019-08-01:
Hi All.

Thanks for the question. I have reached out to the SDV team to get a good answer for you and will post something once I hear back. Sorry for the delay.

Eliot

dwkaiser, 2019-09-03:
Any update on this issue?

EliotSeattle, 2019-09-09:
Sorry. Still no response. Have you tried posting the question in the [Windows Hardware WDK and  driver development forum](https://social.msdn.microsoft.com/Forums/windowshardware/en-US/home?forum=wdk),  or the [OSR NTDEV forum](http://www.osronline.com/cf.cfm?PageURL=showlists.cfm?list=NTDEV)?

I'll continue to ask the team as well.

Apologies,
Eliot

dwkaiser, 2019-10-08:
I will copy here the response from Sebastian Lerner at the following URL (https://social.msdn.microsoft.com/Forums/windowsdesktop/en-US/fdb9c6db-50d5-41b5-8207-aec4601aebd8/sdv-and-microsoft-class-framework-libraries?forum=wdk):

Hi Dave,
Unfortunately, this is a limitation to SDV that does not have a work-around.  However, there are some generic rules that can still be used such as the nullcheck rule (https://docs.microsoft.com/en-us/windows-hardware/drivers/devtest/nullcheck) to perform some level of static verification.
SDV has not been open sourced and there is no plan to do so at this moment.
To certify for Windows Server, SDV must be run and all "must-fix" errors must be fixed (https://docs.microsoft.com/en-us/windows-hardware/test/hlk/testref/windows-server-must-fix-errors)
For Windows Desktop, there is no requirement to run SDV at this time.

Proposed as answer by Sebastian Lerner (MSFT) Wednesday, October 2, 2019 5:42 PM 
Marked as answer by Brian CatlinModerator Wednesday, October 2, 2019 6:32 PM 
Wednesday, October 2, 2019 5:42 PM

EliotSeattle, 2019-10-09:
@dwkaiser 
Thanks so much, Dave! I appreciate you hunting this down.
I've added the info to the page and have pushed it to the master branch. It's live.
Eliot

closed 2019-10-09 by EliotSeattle