MicrosoftDocs/windows-driver-docs #1691: EhStorCertMgr*.*

riback, 2019-07-29:
I installed WDK, SDK, ADK, visualstudio on Windows 10 1809 and 1903 and could not find any files named EhStorCertMgr*.*. 
I then downloaded the appliance and mounted it - still no sign of EhStorCertMgr*.* files anywhere.
Can you please just make these files available for download, perhaps even on GitHub in this project?

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: f5c83b7e-21ab-120e-b3a6-db9d8a2c0032
* Version Independent ID: a6f33e69-564a-7206-2dcb-f50ad8480cd1
* Content: [Installation of the Enhanced Storage Certificate Management Tool - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/devtest/installation-of-the-enhanced-storage-certificate-management-tool#feedback)
* Content Source: [windows-driver-docs-pr/devtest/installation-of-the-enhanced-storage-certificate-management-tool.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/devtest/installation-of-the-enhanced-storage-certificate-management-tool.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @EliotSeattle
* Microsoft Alias: **eliotgra**

EliotSeattle, 2019-08-14:
Looking into it.

riback, 2019-09-19:
Hi Eliot,
Thanks for looking into these missing files.

Any updates?

ER

DariusCorvus, 2020-02-24:
Same problem, waiting for a Answer...

riback, 2020-04-21:
I found https://github.com/Aekras1a/Labs/tree/master/Labs/WinDDK/7600.16385.1/tools/EnhancedStorage which has files that purport to be the EhStorCert files for x86, x64, ia64. 
Unfortunately, so far I have not found how/where to install them. All of my attempts result in an error message claiming OS version incompatibility. I've tried each of the 3 versions of files in c:\windows\system32 and c:\windows\syswo64

Eliot, you may be our last best hope

ER

EliotSeattle, 2020-04-21:
Uh-oh, my light saber's in the shop. 

@riback @DariusCorvus I am still working on this mystery. Just yesterday I dug up contact info for people who worked on the tool some time ago and am trying to get some history from them. I will keep looking until I get some idea about what happened.

Sorry this is taking so long.
Eliot

EliotSeattle, 2020-10-20:
I am afraid that the tool was deprecated and removed from the kit in 2017. 

I will move this content to our Previous Versions area.

Eliot

closed 2020-10-20 by EliotSeattle