MicrosoftDocs/windows-driver-docs #1698: 笔误？

kouzhudong, 2019-08-01:
FWPS_LAYER_INBOUND_MAC_FRAME_ETHERNET | The beginning of the IP header.
FWPS_LAYER_OUTBOUND_MAC_FRAME_ETHERNET | The beginning of the MAC header.
FWPS_LAYER_INBOUND_MAC_FRAME_NATIVE | The beginning of the MAC header.
FWPS_LAYER_OUTBOUND_MAC_FRAME_NATIVE | The beginning of the MAC header.

IP header？
应该是MAC header吧！
笔误吧！
没有测试验证。

EliotSeattle, 2019-08-01:
Hi @kouzhudong 
Thanks for the feedback. Can you please tell us which page(s) you see this mistake on?
Thanks,
Eliot

kouzhudong, 2019-08-02:
https://docs.microsoft.com/en-us/windows-hardware/drivers/network/data-offset-positions

EliotSeattle, 2019-08-02:
Thanks, @kouzhudong!!
@duncanmacmichael, can you please take a look at this?

duncanmacmichael, 2019-08-02:
Hi @kouzhudong, can you clarify exactly what you're asking here? I translated your question to English and got something like the following:

It's MAC header!
Wrong pen! There is no test validation.

Sorry, I don't know what you mean. In issue [529](https://github.com/MicrosoftDocs/windows-driver-docs/issues/529), I did get a question about this last year, and I got confirmation from a senior engineer that the doc is now correct. To reiterate:

Inbound:
Native – offset is at the MAC Header
Ethernet – offset is at the IP Header
Outbound:
Native – offset is at the MAC Header
Ethernet – offset is at the MAC Header

duncanmacmichael, 2019-08-08:
Hi @kouzhudong, could you please let us know if you still think there's an issue with the documentation here? If not, I'll close this issue soon since I believe the doc is correct.

EliotSeattle, 2019-09-09:
No response for 30 days, so we're going to close this. Feel free to reopen with the requested info.
Thanks!

closed 2019-09-09 by EliotSeattle

kouzhudong, 2019-09-10:
我也让你们知道下，长久不回复的感觉。