MicrosoftDocs/windows-driver-docs #1700: What are the MBIM messages to perform the operation for a Consumer eUICC?

lchen1234, 2019-08-02:
Hi Microsoft Experts,

As you know, for a Consumer eUICC, user requires to support following interaction with eUICC :
- Profile downloading and installation
- Enable a profile
- Disable a profile
- Delete a profile
- eUICC memory reset ( to delete all profiles)
- Profile query list
- Nicknaming

I don't see Microsoft defines any dedicated MBIM CIDs for these operations, anything I missed? Or these operation can be done by the CIDs present in this page? please help clarify?

Thanks

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: c17a35ca-60a6-fd9f-e171-8413f8ea4604
* Version Independent ID: 7516b9db-e539-19d7-9be6-1140f6141404
* Content: [MB low level UICC access - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/network/mb-low-level-uicc-access#feedback)
* Content Source: [windows-driver-docs-pr/network/mb-low-level-uicc-access.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/network/mb-low-level-uicc-access.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @duncanmacmichael
* Microsoft Alias: **dumacmic**

duncanmacmichael, 2019-08-02:
Hi @lchen1234, good question. Between this page and the [MB UICC application and file system access](https://docs.microsoft.com/en-us/windows-hardware/drivers/network/mb-uicc-application-and-file-system-access) topic, that's what we currently have to offer for MBIM information related to UICC operations. Let me see what I can find out about your eUICC questions.

duncanmacmichael, 2019-08-08:
Hi @lchen1234, thanks for waiting. I've gotten an answer from the MBB team, which I'll summarize here:

The CIDs that we expose at the MBIM level are for facilitating Windows's low-level (APDU-level) communication with the eUICC module, meaning he logical operations you're looking for here are not supported at this low-level interface. Windows with eUICC support is shipped with built-in local profile assistant (LPA) functionality for these types of operations.

Basically, programming at this level is not appropriate for these user interatctions; that is handled by higher-level functionality that is built in to Windows. Therefore, there is nothing to document here as far as MBIM CIDs.

Hope this helps.

closed 2019-08-08 by duncanmacmichael