MicrosoftDocs/windows-driver-docs #1703: Show Hidden Devices has more than just info a developer would need

mfstanton, 2019-08-02:
[Enter feedback here] The "Show Hidden Devices tab has a special use for devices that are malfunctioning. In my case I have a faulty disk drive that one device link comes up as okay and there is a second disk drive that shows up with a slightly different ID field that actually has the error information and is Hidden unless requested. I have seen this with multiple devices of the storage variety. In my case there is a string of additional characters that is appended to the TRUE device not so friendly name that gives a subtype for the rev of the version and it is embedded in the ID firmware of the device I am in contact with the manufacturer of this violation of the interface specification.

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 4db97ac5-9a25-c7ad-b054-797c2a00c056
* Version Independent ID: 2b72ce14-6a89-e40b-0e10-651172a1d51a
* Content: [Viewing hidden devices - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/install/viewing-hidden-devices#feedback)
* Content Source: [windows-driver-docs-pr/install/viewing-hidden-devices.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/install/viewing-hidden-devices.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @tedhudek
* Microsoft Alias: **tedhudek**

tedhudek, 2019-09-23:
Hi @mfstanton, thanks for the report.  I'm not clear what your desired action is for the docs though.  What specifically are you recommending for the doc page in question?

mfstanton, 2019-09-24:
Thanks for the feedback Ted. The emphasis on use by a developer is misplaced. Administrators and even power users would find the information useful.  It would help if there was more info available too links to other documents about the information and more explanation of the information available would help. I am an experienced device driver designer/developer but most of my work was on Linux and mainframes.  Less knowledgeable people would find additional details useful if they were easier to find.

tedhudek, 2019-10-18:
Hi @mfstanton, I appreciate the feedback.  If you like, I'd encourage you to submit a pull request if there are particular points/links you'd like to see added to this page.  To do that, click the **Edit** button on the docs.microsoft.com page, and then click the pencil ("edit this file") icon on GitHub.  Happy to take a look at anything you think would be worthwhile.

closed 2019-10-18 by tedhudek