MicrosoftDocs/windows-driver-docs #1709: 2016 OS Version

srveerla, 2019-08-07:
Inf2Cat fails to recognize 2016 OS version.

Inf2Cat.exe /driver:&lt;INF DIR&gt; /os:SERVER2016_X64
Inf2Cat.exe /driver:&lt;INF DIR&gt; /os:Server10_X64

Both the above fail with error "Operating systems parameter invalid."

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 0afbc04b-349a-57ed-eeb0-3d2cd28e359c
* Version Independent ID: de61f684-31d7-2de5-d66c-a3319f5198d9
* Content: [Inf2Cat - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/devtest/inf2cat#feedback)
* Content Source: [windows-driver-docs-pr/devtest/inf2cat.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/devtest/inf2cat.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @EliotSeattle
* Microsoft Alias: **eliotgra**

EliotSeattle, 2019-08-08:
Hi @srveerla 

I am looking into this. I believe that the SERVER2016_X64 string is correct, so I am checking with the product team. Hopefully, I will find out soon if this is a doc bug or some other issue.

Thanks,

Eliot

srveerla, 2019-08-09:
Thanks Eliot. Meanwhile you get the solution, i have couple of other questions.
1. Is this something needs to be fixed from our driver end?
2 Is this something that needs to be mentioned INF file?

Thanks,
Sagar

EliotSeattle, 2019-08-09:
Hi @srveerla 

Which version of Infcat are you using? The team says the version we use in the Hardware Dev Center recognizes those OSes properly:

```
C:\Inf2Cat>Inf2Cat.exe /driver:C:\driver /os:SERVER2016_X64
.........................
Signability test complete.
Errors:
None
Warnings:
None

Catalog generation complete.
C:\driver\delta.cat

C:\Inf2Cat>Inf2Cat.exe /driver:C:\driver /os:SERVER10_X64
.........................
Signability test complete.
Errors:
None
Warnings:
None
Catalog generation complete.
C:\driver\delta.cat
```
Thanks,
Eliot

srveerla, 2019-08-09:
Tool is downloaded from Hardware Dev Center, below are version details:

c:\Program Files (x86)\Microsoft Winqual Submission Tool>Inf2Cat.exe
Version 3.2.0.0
Runs driver signability tests and creates the catalog(s).

INF2CAT /driver:path /os:operatingSystem1[,os2]...
        [/nocat] [/verbose]
        [/drm[:file1[,file2]...]]
        [/pe[:file1[,file2]...]]
        [/pageHashes[:file1][,file2]...]]

  /driver (/drv)    Indicates the path to the driver package follows.

  path              Specifies the path to the driver package.

  /os               Indicates the operating system(s) targeted by the driver
                    package follows. The targeted operating system(s) is a
                    comma separated list of the following values:

  operatingSystem1  2000

                    XP_X86      Server2003_X86
                    XP_X64      Server2003_X64
                                Server2003_IA64

                    Vista_X86   Server2008_X86
                    Vista_X64   Server2008_X64
                                Server2008_IA64

                    7_X86
                    7_X64       Server2008R2_X64
                                Server2008R2_IA64

                    8_X86
                    8_X64       Server8_X64
                    8_ARM

                    6_3_X86
                    6_3_X64     Server6_3_X64
                    6_3_ARM

From the above list, i see the SERVER2016_X64 and SERVER10_X64 are not listed. How to get the latest version of Submission tool?

srveerla, 2019-08-20:
Hi Eliot.

Any update on this?

EliotSeattle, 2019-08-24:
Hi @srveerla 

I am still trying to get an answer about how to download the latest version that does recognize those. Sorry that this is taking so long.

Eliot

tedhudek, 2021-01-21:
Sorry it took us a while - latest WDK has the most recent version that should fix this.

You can find the download here --

https://docs.microsoft.com/en-us/windows-hardware/drivers/download-the-wdk

Thanks for the report and best of luck!

closed 2021-01-21 by tedhudek