MicrosoftDocs/windows-driver-docs #1711: GenericUSBFN with Windows IoT

keshavagn, 2019-08-08:
Whether GenericUSBFN is tested either with Windows or Windows IoT? Can someone please confirm? This is because, in this link: https://social.msdn.microsoft.com/Forums/windowsdesktop/en-US/a79e1e4a-abf3-4ff8-9be5-0fd15ed9ad0a/genericusbfnsys-host-configurations?forum=wdk#09cf3f1c-ff31-43c6-9036-ab784e495d00 it looks like there's some risk here.

duncanmacmichael, 2019-08-08:
Hi @keshavagn, the USB team is currently heads down in development for the next Windows release but they should be free to help answer this for us very soon. I'll let you know when I'm able to connect with them and shed some light on this. Thanks!

tedhudek, 2021-02-24:
Thanks for the feedback and apologies for the delay.

I'm actually not sure.  If this is still an issue, you may have success posting your question in the following forums, where other driver developers and Microsoft engineers can be found. The community there is quite responsive.

* [Windows Hardware WDK and Driver Development](https://social.msdn.microsoft.com/Forums/windowsdesktop/en-US/home?forum=wdk)
* [OSR's ntdev List](http://www.osronline.com/showlists.cfm?list=ntdev)

Again, thanks for reaching out. Best of luck with this issue.

closed 2021-02-24 by tedhudek