MicrosoftDocs/windows-driver-docs #1715: .shell has terrible output

ShenXiaolong1976, 2019-08-13:
for example :
0:000&gt; .shell -ci "*" echo %WinScriptPath:\=\\%
&lt;.shell waiting 10 second(s) for process&gt;
D:\\work\\shenxiaolong\\core\\WinScript
.shell: Process exited

sometimes, it outputs:
0:000&gt; .shell -ci "*" echo %WinScriptPath:\=\\%
D:\\work\\shenxiaolong\\core\\WinScript
.shell: Process exited

I need to parse the output as below command: (here WinScriptPath is system environment variable)
.foreach /ps 3 (var { .shell -ci "*" echo %WinScriptPath:\=\\%} ) {.printf /D "&lt;exec cmd=\".cmdtree ${var}\\Windbg\\setting\\cmdTree.dbg\"&gt;open cmd panel&lt;/exec&gt;\r\n"}  

but above non-fixed output string disturb my parse.
sometimes output includes below string, but sometimes not, it should be one bug if not option disables it.
&lt;.shell waiting 10 second(s) for process&gt;
so I have to use ".foreach /ps 3" sometimes, and or use ".foreach /pS 6  /ps 3" sometimes and it is not detected.

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: c96c0422-a4e8-e634-2824-a1a39e44f5ce
* Version Independent ID: 7b0cbd65-ba31-4688-f399-7cb532c58dba
* Content: [.shell (Command Shell) - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/debugger/-shell--command-shell-)
* Content Source: [windows-driver-docs-pr/debugger/-shell--command-shell-.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/debugger/-shell--command-shell-.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @DOMARS
* Microsoft Alias: **domars**

ShenXiaolong1976, 2019-08-13:
I have one suite of windbg script and external shell command script to assist me in analysis and debug dump file or live-debug. 
more than 100 script files, they interact with each other with in/out parameter. those scripts deepen my windbg debug and save me lots of time.

but the .shell output disturbs my script parses result recently.
(it occurs only recently, those script files have worked fine for a long time).
please help, thanks.

rodwiddowson, 2019-08-13:
Did you mean to put this [here](https://github.com/microsoftfeedback/WinDbg-Feedback)?

ShenXiaolong1976, 2019-08-14:
here is the feedback link from MS doc : (google windbg .shell)
https://docs.microsoft.com/en-us/windows-hardware/drivers/debugger/-shell--command-shell-

DOMARS, 2019-08-23:
@ShenXiaolong1976 - This area is for issues on the docs. To provide input on the command output, please post here: https://github.com/microsoftfeedback/WinDbg-Feedback/issues

Longer term you might explore using the debugger data model, although it takes a bit of work to convert to that interface.

closed 2019-08-23 by DOMARS