MicrosoftDocs/windows-driver-docs #1721: Recommendation to support obsolete OIDs

I-gor-C, 2019-08-14:
Most of the OIDs mentioned on this page as recommended for support are obsolete/not sent in NDIS 6.

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 1122be9d-59f5-dadb-4793-224fae0ee683
* Version Independent ID: da4c40cc-197c-0a44-12b9-29c44bdece79
* Content: [NDIS general statistics OIDs - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/network/ndis-general-statistics-oids)
* Content Source: [windows-driver-docs-pr/network/ndis-general-statistics-oids.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/network/ndis-general-statistics-oids.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @duncanmacmichael
* Microsoft Alias: **dumacmic**

duncanmacmichael, 2019-08-15:
Hey @I-gor-C, thanks for the observation. While this is true that many of them are obsolete or are now handled by NDIS in NDIS 6.0+, we do still have to document each of these OIDs given that they ship publicly. This is for both internal knowledge and to avoid confusion with driver writers (who might otherwise think they have to support every one of them, for example). I'll go ahead and close this issue, but if you have any other specific suggestions please feel free to let us know. Thanks!

closed 2019-08-15 by duncanmacmichael

I-gor-C, 2019-08-15:
Thanks @duncanmacmichael. Just a small follow up on this - when the documentation states that a specific OID is obsolete, is it absolutely safe for the miniport driver to assume the OID won't arrive and it's fine not to have a handler for it?

duncanmacmichael, 2019-08-16:
Yes, it should be @I-gor-C. You shouldn't receive those OIDS at all, but if for some weird reason you do then I'd recommend the following:

1. Complete the OID with a not supported status, making sure not to leak the OID.
2. Open a doc bug on that particular OID's page to let us know you received it when you weren't supposed to, so I can pass that info along to the NDIS team and update the doc accordingly.

Hope this helps!