MicrosoftDocs/windows-driver-docs #1723: Sample refers to "ext-ms-win-session-wtsapi32-l1-1-0"

asklar, 2019-08-16:
The sample says:
if (!IsApiSetImplemented("ext-ms-win-session-wtsapi32-l1-1-0"))

However ext-ms-win-session-wtsapi32-l1-1-0 is not in the list of APISets (in fact there are no ext-* entries). Is there a public list of ext- apisets?

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: a013fd99-11c9-aa72-fa0f-2b5f4951fec4
* Version Independent ID: 9e9ff0ea-e13a-1f50-9fe5-9210e037c959
* Content: [Building for OneCore - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/develop/building-for-onecore)
* Content Source: [windows-driver-docs-pr/develop/building-for-onecore.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/develop/building-for-onecore.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @tedhudek
* Microsoft Alias: **tedhudek**

tedhudek, 2019-10-25:
Hi @asklar, sorry for the delay.  When you say that set is not in the list of APISets, do you mean the list at https://docs.microsoft.com/en-us/windows/win32/apiindex/umbrella-lib-onecoreuap?

asklar, 2019-10-26:
@tedhudek  Sorry I'm not following the question. What I said was:

> The sample says:
> if (!IsApiSetImplemented("ext-ms-win-session-wtsapi32-l1-1-0"))
> 
> However ext-ms-win-session-wtsapi32-l1-1-0 is not in the list of APISets (in fact there are no ext-* entries). Is there a public list of ext- apisets?
> 

The sample in question is https://docs.microsoft.com/en-us/windows-hardware/drivers/develop/building-for-onecore

tedhudek, 2019-10-28:
Thanks @asklar.  What I meant was, when you say `this library is not in the list of APISet libraries`, did you mean the list that is at the link I have above?  (which you can get to by clicking **IsApiSetImplemented** from the section you referenced, and then clicking the `OneCoreUap umbrella library` link from there)

tedhudek, 2019-11-04:
Update on this one, @asklar.  We've opened an internal tracking item to add functionality that will display the relevant APISet on the reference page for each API.  In the meantime, if you're blocked because you can't find one of these, let me know and I can try to get an answer for you.

Thanks again for the feedback.

closed 2019-11-04 by tedhudek