MicrosoftDocs/windows-driver-docs #1735: Using extensions in WinDbg Preview is not documented

cbiesinger, 2019-08-25:
It would be nice if somewhere on the WinDbg Preview page, the default extension search path was documented. Apparently it's %LOCALAPPDATA%\Dbg\EngineExtensions

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: c1e093f9-7df0-17ab-71ae-8af9003ee041
* Version Independent ID: 55ed6fd3-a7a6-f7d5-2a8e-fbf6cc6e2c7b
* Content: [Debugging Using WinDbg Preview - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/debugger/debugging-using-windbg-preview#feedback)
* Content Source: [windows-driver-docs-pr/debugger/debugging-using-windbg-preview.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/debugger/debugging-using-windbg-preview.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @DOMARS
* Microsoft Alias: **domars**

DOMARS, 2019-10-25:
@cbiesinger - As it turns out, the ability to auto load extensions is something that is being requested in this WinDbg issue - "Add a way to autoload extensions"

https://github.com/microsoftfeedback/WinDbg-Feedback/issues/32

That feature will need to be added to the product. When that is available, I will document it.

Thanks for using the debugger and providing input on this feature. 

@aluhrs13 - FYI

closed 2019-10-25 by DOMARS

cbiesinger, 2019-10-28:
@DOMARS I'm confused. I'm not requesting a new feature. I'm not asking for auto-loading extensions.

All I was asking for is documenting the existing feature that WinDbg will search for extensions in  %LOCALAPPDATA%\Dbg\EngineExtensions, so if you put a DLL there you can then do "!load foo" without specifying a full path.

DOMARS, 2019-10-28:
@cbiesinger - That path may work, but my understanding is that not a supported feature at this point.

cbiesinger, 2019-10-28:
Oh, ok. Thanks for clarifying that.