MicrosoftDocs/windows-driver-docs #1736: PageProtection is not one of memory protection constant values.

kouzhudong, 2019-08-26:
PageProtection 看定义是位与，其实，在Windows 10上测试，procmon.exe发现它的值会有：PAGE_EXECUTE|PAGE_NOCACHE。

错了，都修正，不要误导人，这些微软的人。

https://correy.webs.com

[Enter feedback here]

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 20a64a62-c8b8-d3f7-942a-145b8d8907c9
* Version Independent ID: 3ecdda0f-ef9b-5bde-431e-81a7af013f64
* Content: [FLT_PARAMETERS for IRP_MJ_ACQUIRE_FOR_SECTION_SYNCHRONIZATION union - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/ifs/flt-parameters-for-irp-mj-acquire-for-section-synchronization#feedback)
* Content Source: [windows-driver-docs-pr/ifs/flt-parameters-for-irp-mj-acquire-for-section-synchronization.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/ifs/flt-parameters-for-irp-mj-acquire-for-section-synchronization.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @lorihollasch
* Microsoft Alias: **loriwhip**

kouzhudong, 2019-09-14:
\Windows Kits\10\Include\10.0.18362.0\km\wdm.h

#define PAGE_NOACCESS           0x01    
#define PAGE_READONLY           0x02    
#define PAGE_READWRITE          0x04    
#define PAGE_WRITECOPY          0x08    

#define PAGE_EXECUTE            0x10    
#define PAGE_EXECUTE_READ       0x20    
#define PAGE_EXECUTE_READWRITE  0x40    
#define PAGE_EXECUTE_WRITECOPY  0x80    

#define PAGE_GUARD             0x100    
#define PAGE_NOCACHE           0x200    
#define PAGE_WRITECOMBINE      0x400    

#define PAGE_GRAPHICS_NOACCESS           0x0800    
#define PAGE_GRAPHICS_READONLY           0x1000    
#define PAGE_GRAPHICS_READWRITE          0x2000    
#define PAGE_GRAPHICS_EXECUTE            0x4000    
#define PAGE_GRAPHICS_EXECUTE_READ       0x8000    
#define PAGE_GRAPHICS_EXECUTE_READWRITE 0x10000    
#define PAGE_GRAPHICS_COHERENT          0x20000    

//
// PAGE_REVERT_TO_FILE_MAP can be combined with other protection
// values to specify to VirtualProtect that the argument range
// should be reverted to point back to the backing file.  This
// means the contents of any private (copy on write) pages in the
// range will be discarded.  Any reverted pages that were locked
// into the working set are unlocked as well.
//

#define PAGE_ENCLAVE_THREAD_CONTROL 0x80000000  
#define PAGE_REVERT_TO_FILE_MAP     0x80000000  
#define PAGE_TARGETS_NO_UPDATE      0x40000000  
#define PAGE_TARGETS_INVALID        0x40000000  
#define PAGE_ENCLAVE_UNVALIDATED    0x20000000  
#define PAGE_ENCLAVE_NO_CHANGE      0x20000000
#define PAGE_ENCLAVE_DECOMMIT       0x10000000

lorihollasch, 2019-09-17:
@kouzhudong, PageProtection is a member of the AcquireForSectionSynchronization structure, and not one of the PAGE_XXX constants. Per PageProtection's description on the refpage, it takes one of the PAGE_XXXX constants (along with possible PAGE_XXX modifiers). 
That said, I see that we are missing documentation for some of the PAGE_XXX values that you copied from wdm.h - thanks for alerting us to this! I've filed an internal task to add the missing values and am going to close this issue.

closed 2019-09-17 by lorihollasch