MicrosoftDocs/windows-driver-docs #1751: Probably wrong documentation

shaharlux, 2019-09-03:
Hi, I think that what documented as parameter 2 above is actually parameter 4 while parameter 2 is Reserved

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 4a1ce3a1-3126-61d3-fc20-59386092330d
* Version Independent ID: d24c7707-8bcc-57d2-0427-21b0efbbf5f7
* Content: [Bug Check 0x93 INVALID_KERNEL_HANDLE - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/debugger/bug-check-0x93--invalid-kernel-handle)
* Content Source: [windows-driver-docs-pr/debugger/bug-check-0x93--invalid-kernel-handle.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/debugger/bug-check-0x93--invalid-kernel-handle.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @DOMARS
* Microsoft Alias: **domars**

DOMARS, 2019-09-06:
@shaharlux - I'm investigating. It looks like if there is a one present in parameter 4, that is a different situation then what is currently documented , where the parameters will need to be treated differently. I will dig further with the dev team. If parameter 4 is zero, then the docs are OK as is. Thanks for pointing what you are seeing with parameter 4.

DOMARS, 2019-10-10:
@shaharlux - Thanks for this issue and your interest in the Windows stop code docs. I worked with the dev team to update this stop code to update this topic to match current Windows behavior.

closed 2019-10-10 by DOMARS