MicrosoftDocs/windows-driver-docs #1760: Situations which require full file system drivers include "migrating data from other OS disks"

kyanha, 2019-09-08:
There exist valid situations where full filesystem drivers must be created.  These include access to ext3 (Linux), HFS+ (MacOS), APFS (MacOS), AFS (multi-platform), zfs, the filesystems used by OpenBSD and FreeBSD, and many others -- as well as kernel reflectors to user-mode file system implementations, such as Dokany.  The situation can best be described as "when the on-disk format one is trying to interact with is not handled by a currently shipping driver".

A full discussion of the situations which might require a full file system driver to be written would be very much appreciated.

(Also, there's precious little expression of why a filter driver would be more appropriate than a minifilter driver.  Explaining *why* decisions should be approached as Microsoft recommends, as well as the pros and cons of each option, would be very helpful.  There's such a discussion in the DDK documentation about choosing which driver model to support.  As a side note, it might be desirable to have that discussion in any place where Microsoft has obsoleted an older technology in favor of a new technology, but still supports the older one.)

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 19242c27-5e25-bd42-7cf4-4e80220ed690
* Version Independent ID: 08880e36-6f5e-b89c-571e-968c1307d6fa
* Content: [File System Drivers - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/ifs/file-system-drivers)
* Content Source: [windows-driver-docs-pr/ifs/file-system-drivers.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/ifs/file-system-drivers.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @lorihollasch
* Microsoft Alias: **loriwhip**

kyanha, 2019-09-13:
I was just looking through the Windows Developer Feedback site, and I found this (which indicates at the very least a need for information about this, if Microsoft doesn't want to do the file system development work itself): https://wpdev.uservoice.com/forums/266908-command-prompt-console-windows-subsystem-for-l/suggestions/13818042-support-mounting-xfs-ext2-ext3-ext4-etc-partition

lorihollasch, 2020-03-10:
@kyanha, thanks also for this thoughtful PR and apologies again for my slow response  -- I wrapped the documentation updates into the TOC reorganization (per your ECP report in #1759 ). Given the myriad complexity of reasons for why one might need to implement a FS, given that this is so rarely necessary, and given that a sample FS is available as a starting point for anyone choosing to develop their own, the dev team is not able to provide FS development guidance. I have updated the guide to make clear that sample code is available. I've also added information that steers filter developers to the minifilter model (adding the word "legacy" in the TOC for legacy filter info, along with more "NOTE: use minifilter model" language on legacy pages). No one should be developing legacy filters anymore. Thanks again!

closed 2020-03-10 by lorihollasch