MicrosoftDocs/windows-driver-docs #1762: The table is broken

ChuckM2, 2019-09-10:
The table on this page is broken and not readable.

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: a4806b83-6531-a735-77d3-efa6daf8fee5
* Version Independent ID: e77c4728-0c71-67b7-59a6-b5e1cc9180b7
* Content: [Details of the Extended Format - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/display/details-of-the-extended-format#feedback)
* Content Source: [windows-driver-docs-pr/display/details-of-the-extended-format.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/display/details-of-the-extended-format.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @lorihollasch
* Microsoft Alias: **loriwhip**

EliotSeattle, 2019-09-10:
Thanks, @ChuckM2 

That's awful. I'll see if we can't get to that today.

ChuckM2, 2019-09-12:
The table is looking much better. Thank you.

closed 2019-09-12 by EliotSeattle