MicrosoftDocs/windows-driver-docs #1771: Short-Sighted Policy Change

timrprobocom, 2019-09-18:
This policy would work great in an environment where every system runs Windows 10 (like in Redmond), where attestation signing works.  In the Real World, MANY of us are still publishing new driver packages for WIndows 7, 8, and 8.1, where attestation signing is not applicable and the cross-signing technique is the ONLY alternative.  It's just not acceptable to require WHQL for these old systems.  There are too many driver classes where WHQL doesn't work.  I assume this policy will be rescinded after an industry outcry.  You can't make major policy changes like this in a bubble.

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 0e326d3e-833a-e312-d35d-11569690c7e6
* Version Independent ID: 7a9795cf-093e-a67f-8162-b832e1161ba8
* Content: [Deprecation of Software Publisher Certificates, Commercial Release Certificates, and Commercial Test Certificates - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/install/deprecation-of-software-publisher-certificates-and-commercial-release-certificates#feedback)
* Content Source: [windows-driver-docs-pr/install/deprecation-of-software-publisher-certificates-and-commercial-release-certificates.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/install/deprecation-of-software-publisher-certificates-and-commercial-release-certificates.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @tedhudek
* Microsoft Alias: **tedhudek**

nasantia, 2019-09-20:
Hi Tim,
I'm Nate from Microsoft's Trusted Root Program, and we're the team responsible for this new policy. 

First question I want to check in on is whether opening up Attestation signing down to Windows 7 would resolve the majority of the issues? And if not, what are some use cases where WHQL nor Attestation would suffice? 

And Second, you mentioned several driver classes where it doesn't make sense to go through the full WHQL process. What are some examples of these driver classes?

timrprobocom, 2019-09-20:
On Sep 19, 2019, at 7:53 PM, nasantia <notifications@github.com> wrote:
> 
> I'm Nate from Microsoft's Trusted Root Program, and we're the team responsible for this new policy.
> 
I’m glad you wrote.  The guys at OSR are also checking into this.  Microsoft keeps on making it harder and harder for driver writers to do their job.  This just seems like a totally unnecessary additional burden on us.  We all understand you want to pretend that nothing older than Windows 10 exists, but it’s just not true.  ALL of my customers still want Windows 7, 8, and 8.1, and a few still ask for Vista and XP.

> First question I want to check in on is whether opening up Attestation signing down to Windows 7 would resolve the majority of the issues? And if not, what are some use cases where WHQL nor Attestation would suffice?
> 
Having attestation signing for the older systems would solve most of the cases, but certainly not all.  When I’m doing testing, I want to have my system in essentially the same state as my end users.  That means I’m not going to run “test mode”.  If I can’t use cross-signing, that means I have to get attestation signing for every test driver I build.  Right now, there’s no easy path for automating attestation signing as part of the build process, and even if there were, it’s a very lengthy process (compared to build times).

And for legacy drivers that don’t need an INF, the attestation process is just awkward, and of course there is no WHQL path..  Attestation can be done by creating a package with a fake INF, but that just seems like a terrible hack, doesn’t it?

> And Second, you mentioned several driver classes where it doesn't make sense to go through the full WHQL process. What are some examples of these driver classes?
> 
People are extremely creative.  Many folks are creating devices and drivers that just don’t fall neatly into existing driver categories, and no matter how much you try to keep up, people are ALWAYS going to stay ahead of you.  It used to be easy to choose the “unclassified” category, but the HLK tries to automate all of that, and it’s not always right.

Plus, I’m sure you’re aware that a full HLK run is a very lengthy process.  No one is going to go through that for anything other than driver packages that are definitely being shipped to the outside world.  And, as I said, WHQL doesn’t do anything for non-INF drivers.

I assume you created and instituted this policy without consulting with your industry partners.  ANY professional driver writer could have told you what bad idea this is.
— 
Tim Roberts, timr@probo.com
Providenza & Boekelheide, Inc.

tedhudek, 2019-10-17:
I'm going to close this one out, in the absence of doc-specific feedback.

Tim, I'll leave it to you and Nate to connect further in a different space as needed to continue the conversation about the policy in question.

closed 2019-10-17 by tedhudek