MicrosoftDocs/windows-driver-docs #1788: Migration Issue: characters render with extra backslash

EliotSeattle, 2019-09-24:
First reported in private repo PR https://github.com/MicrosoftDocs/windows-driver-docs-pr/pull/360#issuecomment-534602184

Not sure how we check to see if there are more instances. @tedhudek I am assigning this to you, just for visibility on the larger issue. @billlattimer can you please fix these particular instances?

Each of these files contain one or more instances of a character rendering incorrectly; that is, an extra backslash is included.
Expected: (*)
Actual: (\*)

```In these files:
windows-driver-docs-pr/debugger/bl--breakpoint-list-.md	
   Line 32:     You can use an asterisk (**\\***) to indicate 
   Line 80:     given as three asterisks ("**\*\*\\***"),
   Line 109:   This breakpoint is not a thread-specific breakpoint (**\*\*\*\\***).

windows-driver-docs-pr/debugger/dt--display-type-.md
   Line 143:   ends with an asterisk (**\\***)

windows-driver-docs-pr/debugger/gflags-commands.md
   Line 149:   (substitute for any single character) and **\\*** (substitute for

windows-driver-docs-pr/debugger/http-symbol-stores.md
   Line 92:    For **Extension**, type **\\***.

windows-driver-docs-pr/debugger/symchk-command-line-options.md
   Line 62:   asterisk (**\\**<em>) wildcard is permitted. If *FileNames</em> ends in a slash
   Line 65:   single asterisk ( **\\*** ), SymChk will
   Line 78:   is a single asterisk ( **\\*** ), SymChk
   Line 81:   and an asterisk wildcard (**\\***) is permitted

windows-driver-docs-pr/debugger/using-the-help-documentation.md
   Line 40:   and an asterisk (**\\***) to

windows-driver-docs-pr/debugger/x86-instructions.md
   Line 16:   marked with an asterisk (**\\***) are particularly

windows-driver-docs-pr/devtest/28132-driver-taking-the-size-of-pointer.md
   Line 34:   LPSTR, **char\\*** or even **void\\***) [[both]]	

windows-driver-docs-pr/devtest/devcon-examples.md
   Line 164:   the wildcard character (**\\***) to represent
   Line 172:   the wildcard character (**\\***) to
   Line 192:   the wildcard character (**\\***) to
   Line 362:   the wildcard character (**\\***) to
   Line 690:   the wildcard character (**\\***) to
   Line 817:   the wildcard character (**\\***) to	

windows-driver-docs-pr/devtest/devcon-general-commands.md
   Line 125:   For example, **devcon hwids \*pnp\* \*mou\\***
   Line 129:   For example, **devcon hwids =media \*pnp\* \*microsoft\\***
   Line 147:   a DevCon command (for example, **return = devcon hwids \\***).

windows-driver-docs-pr/network/standardized-inf-keywords-for-network-devices.md
   Line 49:   start with an asterisk (**\\***).

windows-driver-docs-pr/print/gpd-file-entry-format.md
   Line 20:   **\\*** <em>EntryName
```

billlattimer, 2019-09-25:
This appears to be a conversion bug from HTML to Markdown. If there was HTML emphasis on the char then it incorrectly converted '<em>*</em>' to '**\\****' which does not parse correctly. It appears that it was possibly a two-pass approach; add the escape, then convert the tags - this learning could be useful for future bulk conversions. 

Note that there are multiple instances of remaining <em> tags, broken conversions and corrected issues in all of the items I noted, including for other escaped chars like brackets and quotes. 

Editing to a standard escape '\*' or a Markdown emphasized escape '**\***' works.

Not sure if there's a pattern to grep other than perhaps \\ combined with *, ", \ or other escaped chars. I don't think we can bulk convert in debugger due to the unique nature of much of the content, but rather likely need to just "power grep" through the issues.

billlattimer, 2019-09-25:
@tedhudek @EliotSeattle these are all complete and previewed, doing a PR. Will close after verified publish. 

Verified against Don's debugger .CHM to ensure no errors.

tedhudek, 2019-09-25:
Fixed in https://github.com/MicrosoftDocs/windows-driver-docs-pr/pull/363

closed 2019-09-25 by tedhudek