MicrosoftDocs/windows-driver-docs #1796: More detail about Full dump vs mini dump

rskorski, 2019-10-02:
I'd like to know more about the differences between a full dump and a mini dump created with /mf or /ma, maybe something like the /m** option that is equivalent to /mf.  This doc currently lists some of it, but has an ambiguous "And other information" part.

The reason for that is knowing more about the WER registry keys I should be using.  To get memory dumps with that mechanism, the easiest option is to flip between mini and full.  I can also use custom to specify the /m** flags, but then I have to dig up the values of the enum (which I cannot find outside of the .h file on my own dev machine).  If I knew what /f really gets me, I'd feel more confident about which options to use for WER registry keys.

ref:  https://docs.microsoft.com/en-us/windows/win32/wer/collecting-user-mode-dumps

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 1abadaf5-ed0e-26fe-1e05-98a32b39e905
* Version Independent ID: 510626db-4304-8d4f-f4db-b81cbe53ccd1
* Content: [User-Mode Dump Files - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/debugger/user-mode-dump-files#feedback)
* Content Source: [windows-driver-docs-pr/debugger/user-mode-dump-files.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/debugger/user-mode-dump-files.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @DOMARS
* Microsoft Alias: **domars**

DOMARS, 2019-10-04:
@rskorski - Not sure I'm following what additional dump information that would be good to add descriptions for. For mf it has this description for example. 
*Adds full memory data to the minidump. All accessible committed pages owned by the target application will be included.*
It may take more work, but I would lean towards selecting between the mini dump options as that is more granular and has more options.
For guidance on when to select which dump options, I would defer to the OSR community forums if you have a specific scenario.

[OSR's windbg List](https://community.osr.com/categories/windbg)
--

rskorski, 2019-10-04:
I apologize, I meant the /f flag, not /mf (edited inline)

The doc says it includes "other information that will be useful to the debugger."   What other info, exactly?  

This info isn't just for me, I'm sharing this with many other devs that aren't familiar with this type of debugging.  For the most part, folks are going to walk away thinking a full dump is going to include everything, and I'd like to tell them exactly what would be missing with /f compared to /ma.

DOMARS, 2019-10-04:
Hmm. This general text seems to offer guidance on the general differences on mini/full.

The name "minidump" is misleading, because the largest minidump files actually contain more information than the "full" user-mode dump. For example, .dump /mf or .dump /ma will create a larger and more complete file than .dump /f. For this reason, .dump /m[MiniOptions] recommended over .dump /f for all user-mode dump file creation.

Full has this:
This dump file includes the entire memory space of a process, the program's executable image itself, the handle table, and other information that will be useful to the debugger.

And ma has this:
The /ma option is equivalent to /mfFhut -- it adds full memory data, handle data, unloaded module information, basic memory information, and thread time information to the minidump.

So that would seem to allow you to compare and contrast the dump types. 

In addition to the docs, Tariks book "Inside Windows Debugging" might be helpful for additional detail on dump types.

For user mode, I would add an important third "dump" type with a whole another dimension of information described in [Time Travel Debugging - Overview](https://docs.microsoft.com/en-us/windows-hardware/drivers/debugger/time-travel-debugging-overview). This assumes you can reproduce the behavior of interest within a reasonable time frame and capture a time travel trace.

rskorski, 2019-10-04:
I agree there is guidance on the general difference, I'm suggesting that it should be more specific.  More pointedly: I think the mention of "other information" is unnecessarily vague.  As a reader, I would rather specifics on what makes up that other information.

DOMARS, 2019-10-08:
@rskorski - I added this additional wording to the topic:

*... and other information that will be useful to the debugger in reconstructing the memory that was in use when the dump occurred.*

I am aware that this is a small step in what can be done here to add clarity on the details of what is contained in each dump type. So in addition, I have also opened an internal work item against myself to work with the dev team to add more detail as well as validate that the dump descriptions are still accurate for user and kernel mode dumps with the current version of Windows. 

As most teams we are resource constrained, and right now I'm scheduled to work on a new tool as well as the stop code / bug check updates for the next release of Windows.

Thanks for spending the time to explain the specific information that would be beneficial to include on dump types and explaining your use case scenario.

closed 2019-10-08 by DOMARS

adamkimsft, 2021-08-18:
I'm here looking for the same thing.  It would be great to know exactly what /f is equivalent to when using /m***.  The docs say /mf has more info than /f.  So it's hard to see how that can be just by adding the f option to /m when the f option adds full memory that's accessible by the exe which is what I though /f by itself would do.