MicrosoftDocs/windows-driver-docs #1798: Provision a computer for driver deployment problem

w103csh, 2019-10-03:
Using the Driver extension for VS 2019 after installing the Windows Driver SDK I use the configure devices tool. Add New Device -> enter a valid network host name and hit Next.

The deployment fails with this error from the log:

> Progress event: Current: 13, Max: 15, Message: "Configuring kernel debugger settings (possible reboot)"
> [14:10:57:864]: Configuring kernel debugger settings (possible reboot)
> [14:10:57:864]: Removing any existing files from test execution folder.
> [14:10:57:864]: Copying required files for "Configuring kernel debugger settings (possible reboot)".
> [14:10:58:125]: An error occured while deploying files to the target machine for test "Configuring kernel debugger settings (possible reboot)": Could not find a part of the path 'C:\Program Files (x86)\Microsoft Visual Studio\2019\Professional\VC\Redist\MSVC\14.23.27820\debug_nonredist\X64\Microsoft.VC141.DebugCRT'..
> Task result status updated: Fail
> [14:10:58:125]: ERROR: Task "Configuring kernel debugger settings (possible reboot)" failed to complete successfully. Look at the logs in the driver test group explorer for more details on the failure.
> Progress event: Current: 13, Max: 15, Message: "    ERROR: Task "Configuring kernel debugger settings (possible reboot)" failed to complete successfully. Look at the logs in the driver test group explorer for more details on the failure."
> Result completed
> Test process exit code: 0

I believe the path it should be looking for is 'C:\Program Files (x86)\Microsoft Visual Studio\2019\Professional\VC\Redist\MSVC\14.23.27820\debug_nonredist\X64\Microsoft.VC14**2**.DebugCRT'

That is Microsoft.VC14**2**.DebugCRT instead of Microsoft.VC14**1**.DebugCRT. But I am not sure.

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: ce050ea3-d61e-830c-3038-f725eb4c29e1
* Version Independent ID: eb7fc65e-bcfe-dc86-3f8c-336d20eb0c86
* Content: [Provision a computer for driver deployment and testing (WDK 10) - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/gettingstarted/provision-a-target-computer-wdk-8-1)
* Content Source: [windows-driver-docs-pr/gettingstarted/provision-a-target-computer-wdk-8-1.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/gettingstarted/provision-a-target-computer-wdk-8-1.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @EliotSeattle
* Microsoft Alias: **eliotgra**

EliotSeattle, 2019-10-03:
Thanks you, @w103csh We'll look into this. Since this isn't a documentation bug, we'll see if we can't get this in the right hands.
@DOMARS 
Can you please take a quick look at this and determine if this should be routed to the windbg team?
Thanks,
Eliot

corytodd, 2019-10-12:
Looks like a known bug. The directory junction solution posted at https://developercommunity.visualstudio.com/content/problem/596096/visual-studio-2019-wdk10.html works for me. 

Perhaps add one of those nifty green boxes to mention this potential blocker?

DOMARS, 2020-01-30:
@w103csh and @corytodd - I added a green tip box to point over to the WDK forum, where this known issue is described. Thanks for your input and good luck in your development efforts.

closed 2020-01-30 by DOMARS

ycelik, 2020-02-24:
​Here is a simple solution which worked for me (x64):

- navigate to "**C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\VC\Redist\MSVC\14.24.28127\debug_nonredist\x64**"
- copy and paste the VC142 folders here e.g. "**Microsoft.VC142.DebugCRT**" 
- rename the copied folder names to VC141 e.g. "**Microsoft.VC141.DebugCRT**"
- build -> deploy -> no more errors

DOMARS, 2020-02-25:
@ycelik - A similar work around is also described on the top level notes on the WDK forum page.

I repaired the link by adding en-us, so that it takes you to the WDK forum page.  Look for "Hardware development kits for Windows 10, Version 1903" and then "Error seen when Provisioning a computer for driver deployment and testing when using WDK 1903 and Visual Studio 16.1 +"