MicrosoftDocs/windows-driver-docs #1801: Page seems empty

Impulse87, 2019-10-04:
There are no links or further information/overview -&gt; https://prnt.sc/pezbdj

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: d4ed803f-575f-b513-b210-ab308cab3eed
* Version Independent ID: e4a318a0-1490-bc72-4228-1768cd9bed9e
* Content: [Overview of Driver Technologies - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/device-and-driver-technologies#feedback)
* Content Source: [windows-driver-docs-pr/device-and-driver-technologies.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/device-and-driver-technologies.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @windows-driver-content
* Microsoft Alias: **windowsdriverdev**

DOMARS, 2019-10-04:
@EliotSeattle - Should this page list the driver technologies or provide general information on drivers?

EliotSeattle, 2019-10-07:
I've added content so that we steer away from the minimalist view. We now have general information and technology links.

closed 2019-10-07 by EliotSeattle