MicrosoftDocs/windows-driver-docs #1807: No Samples Version for Extend.def

robi-g-itl, 2019-10-10:
Have searched through WDK8, 8.1, and 10 samples, and cannot find a file called 'Extend.def'

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 9d55cadd-9e44-7200-4e65-efd0ca9ead38
* Version Independent ID: 7179ce40-024c-4b8e-3191-4a7ca7798f88
* Content: [Building the Extension Unit Sample Control - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/stream/building-the-extension-unit-sample-control#feedback)
* Content Source: [windows-driver-docs-pr/stream/building-the-extension-unit-sample-control.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/stream/building-the-extension-unit-sample-control.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @barrygolden
* Microsoft Alias: **bagold**

robi-g-itl, 2019-10-10:
Don't worry, didn't realise it was in the WDDK7 samples

barrygolden, 2021-01-12:
Updated Step 6 to "Create Uvcxuplgn.def as follows:".

closed 2021-01-12 by barrygolden