MicrosoftDocs/windows-driver-docs #1808: Correction-Spelling

Beejmoula, 2019-10-11:
[Enter feedback here]

---
#### Document details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 3104beb2-e766-f5e7-49f7-a6488592fb2e
* Version Independent ID: a016fbce-2bc7-c1bb-3035-053981676cac
* Content: [Switch from Legacy MBR disk to GPT disk with Windows 10 - Windows drivers](https://docs.microsoft.com/en-au/windows-hardware/drivers/bringup/switch-from-legacy-mbr-disk-to-gpt-disk-with-windows-10)
* Content Source: [windows-driver-docs-pr/bringup/switch-from-legacy-mbr-disk-to-gpt-disk-with-windows-10.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/bringup/switch-from-legacy-mbr-disk-to-gpt-disk-with-windows-10.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @barrygolden
* Microsoft Alias: **bagold**

EliotSeattle, 2019-10-14:
Hi @Beejmoula 
Thanks for reaching out. Can you point out any specific spelling errors? I don't see any.
Thanks,
Eliot

Camellin, 2019-10-28:
No puedo ver mis notificaciones

closed 2020-10-23 by EliotSeattle