MicrosoftDocs/windows-driver-docs #1816: Selecting debug devices in ACPI DBG2 table or explicit custom KdNet dll

jcbottorff-marvell, 2019-10-15:
I don't see how using BCDEDIT I would select a custom debug device, like say a custom hardware simulator interface, that's specified in the ACPI DBG2 table. I basically would want my custom KdNet DLL to be loaded by the OS as the debug transport. There are samples for writing these in the debugger tree of the SDK. Or is there way to specify an explicit KdNet DLL to be loaded? 

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: bed190a3-8bdc-679a-8326-937bea5bf172
* Version Independent ID: 5cfb21ff-c20d-9ced-0ce8-d1dce6d5538f
* Content: [Boot Parameters to Enable Debugging - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/devtest/boot-parameters-to-enable-debugging#feedback)
* Content Source: [windows-driver-docs-pr/devtest/boot-parameters-to-enable-debugging.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/devtest/boot-parameters-to-enable-debugging.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @EliotSeattle
* Microsoft Alias: **eliotgra**

EliotSeattle, 2019-10-15:
Interesting questions, @jcbottorff-marvell 
@DOMARS IS this something to add to the doc?

jcbottorff-marvell, 2019-10-15:
Seem like a doc bug, if you type 'BCDEdit /? TYPES OSLOADER' among other things you get an entry that says 'DBGTRANSPORT (string)   Defines the file name for a private debugger transport'. The question is does this actually work on current OS builds, is this just leftover help text in bcdedit.

jcbottorff-marvell, 2019-10-15:
Actually, I see typing bcdedit /? TYPES lists a number of categories that each have all kinds of poorly documents options.

DOMARS, 2019-10-25:
Thanks for reaching out to the Windows Driver Developer Documentation team. As replacing the debugger transport is not a common request, and we are resource constrained on the docs team, we will not be able to expand this area of the documentation.

You may have success posting your question in the following forums, where other driver developers and Microsoft engineers can be found. The community there is quite responsive.

[OSR's windbg List](https://community.osr.com/categories/windbg)
[Windows Hardware WDK and Driver Development](https://social.msdn.microsoft.com/Forums/windowsdesktop/en-US/home?forum=wdk)

I would mention two other things that may be helpful. The set dbgtransport command is used to set the transport dll for example:

` bcdedit /set dbgtransport kdhvcom.dll`

As you may be aware there are code samples and a readme.docx directory that describe an EXDI sample that you may be able to extract some information on how that works. I don't know to what degree that would apply to your work.

C:\Program Files (x86)\Windows Kits\10\Debuggers\x86\sdk\samples\exdi\ExdiKdSample

Good luck in your efforts with creating  a custom transport.

closed 2019-10-25 by DOMARS