MicrosoftDocs/windows-driver-docs #1817: This isn't read only (is it)

rodwiddowson, 2019-10-15:
&gt; The OPLOCK_KEY_ECP_CONTEXT structure is read-only. You should use it to retrieve information about the oplock key ECP only.

but also 

&gt; A caller can explicitly provide the oplock key to the IoCreateFileEx routine to create the stream handle. 

Which sort of infers that you slam in an ECP (the IoCreateFileEx makes no mention of this).

Also you need the general macro you must use by now to mean IoCreate/FltCreate and all their subvariants

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 144f6345-d579-ca25-f266-dd5f8a73786c
* Version Independent ID: 483f4725-0536-d954-b7d8-cbed818f605c
* Content: [OPLOCK\_KEY\_ECP\_CONTEXT structure - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/ifs/oplock-key-ecp-context)
* Content Source: [windows-driver-docs-pr/ifs/oplock-key-ecp-context.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/ifs/oplock-key-ecp-context.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @lorihollasch
* Microsoft Alias: **loriwhip**

cgallred, 2019-10-23:
@rodwiddowson the wording is kind of misleading.  It means a filter should not alter the contents of the structure when it sees the ECP coming down from above.  You can specify your own (and indeed should) when issuing your own creates in response to a create that has this ECP, as described in the third paragraph of [System-Defined ECPs](https://docs.microsoft.com/en-us/windows-hardware/drivers/ifs/system-defined-ecps).

To specify an ECP list to `IoCreateFileEx` you put it in the `ExtraCreateParameter` field of the `IO_DRIVER_CREATE_CONTEXT` passed through `IoCreateFileEx`'s `DriverContext` parameter.

BTW, `OPLOCK_KEY_ECP_CONTEXT` is obsolete starting in Windows 8.  Instead filters should use [DUAL_OPLOCK_KEY_ECP_CONTEXT](https://docs.microsoft.com/en-us/windows-hardware/drivers/ifs/dual-oplock-key-ecp-context) (@lorihollasch we need to note that in the documentation).

lorihollasch, 2019-11-09:
@rodwiddowson , thanks for flagging this; and @cgallred , thanks for clarifying. I've updated the page with obsolete information (and also updated it say "should not alter" versus "read only").

closed 2019-11-09 by lorihollasch