MicrosoftDocs/windows-driver-docs #1827: Troubleshooting steps

LuigiBruno, 2019-10-18:
How to troubleshoot this bug check?

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: da3013a5-74bf-503b-5ca7-0091aeb933ee
* Version Independent ID: dcd44cd7-6595-0b76-6b76-9a08bc2823fd
* Content: [Bug Check 0x1CA SYNTHETIC_WATCHDOG_TIMEOUT - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/debugger/bug-check-0x1ca--synthetic-watchdog-timeout)
* Content Source: [windows-driver-docs-pr/debugger/bug-check-0x1ca--synthetic-watchdog-timeout.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/debugger/bug-check-0x1ca--synthetic-watchdog-timeout.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @DOMARS
* Microsoft Alias: **domars**

DOMARS, 2020-02-27:
@LuigiBruno  - Thanks for the input, that we should add additional information on Bug Check 0x1CA. 

We use customer input like this as well as data on how often the bug checks occur, to determine how we allocate our documentation team resources.

DOMARS, 2020-02-27:
@LuigiBruno  - Thanks for the input that we should add additional information on this Bug Check. We use customer input like this as well as data on how often the bug checks occur to determine how we allocate our documentation team resources.

For additional tips on troubleshooting, if you have active issue, I would recommend the forums, for example the WDK one:

•	[Windows Hardware WDK and Driver Development](https://social.msdn.microsoft.com/Forums/windowsdesktop/en-US/home?forum=wdk)

LuigiBruno, 2020-03-04:
> @LuigiBruno - Thanks for the input that we should add additional information on this Bug Check. We use customer input like this as well as data on how often the bug checks occur to determine how we [...]

Thanks @DOMARS.
That is one of the Forums I often come back to.

DOMARS, 2021-01-29:
@LuigiBruno - I added using the !analyze command  as one tool that can help on Bug Check 0x1CA SYNTHETIC_WATCHDOG_TIMEOUT. The watchdog timers are tricky as something else may have stopped that then leads to the Bug Check. A general technique, described in https://docs.microsoft.com/en-us/windows-hardware/drivers/debugger/blue-screen-data that may help here is to take full advantage of anything in the event log that occurs right before this bug check. I have an internal work item to look at longer term, if there are other common techniques that we can describe for the watchdogs.

closed 2021-01-29 by DOMARS

LuigiBruno, 2021-02-22:
> @LuigiBruno - I added using the !analyze command as one tool that can help on Bug Check 0x1CA SYNTHETIC_WATCHDOG_TIMEOUT. The watchdog timers are tricky as something else may have stopped that then [...]

OK @DOMARS.
Thank you.