MicrosoftDocs/windows-driver-docs #1828: Changer Drive info needs specific information or privilege assignments

tolistim, 2019-10-19:
Since the "Backup Operators" group is the logical group for user management of tape and changer devices, there should be explicit documentation on how to define and assign user/group access privileges for these device types.

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 429c7b05-1fe3-e368-b32f-ab4a60e52809
* Version Independent ID: 482d2688-426d-79e8-6830-748e090769bb
* Content: [Changer Drivers - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/storage/changer-drivers#feedback)
* Content Source: [windows-driver-docs-pr/storage/changer-drivers.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/storage/changer-drivers.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @lorihollasch
* Microsoft Alias: **loriwhip**

lorihollasch, 2020-08-17:
@tolistim , thanks for reaching out, and many apologies for my slow response. Access control is actually handled by the [mass-storage driver](https://docs.microsoft.com/en-us/windows-hardware/drivers/storage/the-changer-driver-model). For more information about what this looks like at the driver level, see [CD-ROM Drivers->ACLs and the Device Stack](https://docs.microsoft.com/en-us/windows-hardware/drivers/storage/acls-and-the-device-stack).

closed 2020-08-17 by lorihollasch

tolistim, 2020-08-18:
Hi Lori.

We actually discovered that the mix of vendors involved in supplying tape and medium changer drivers is the real issue.

We determined that by disabling the device and then reenabling it when we launch our app and get a permission denied error sorts this out properly so that "Backup Operators" can then access the devices.

Tim