MicrosoftDocs/windows-driver-docs #1829: win cmd

goodandrewsoft, 2019-10-20:
How do that `bp kernel32!CreateEventW "$$<c:\\commands.txt"` in win cmd ?
like
`windbg -c "" program.exe`
?

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 2c4b58b8-0a24-ce41-30e7-1c8936035555
* Version Independent ID: e1b5bc94-3951-9059-5a76-c12bfb9100a7
* Content: [Conditional breakpoints in WinDbg and other Windows debuggers - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/debugger/setting-a-conditional-breakpoint)
* Content Source: [windows-driver-docs-pr/debugger/setting-a-conditional-breakpoint.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/debugger/setting-a-conditional-breakpoint.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @DOMARS
* Microsoft Alias: **domars**

garion100, 2019-10-20:
> Odesláno: neděle 20. října 2019 14:09
> Komu: MicrosoftDocs/windows-driver-docs <windows-driver-docs@noreply.github.com>
> Předmět: [MicrosoftDocs/windows-driver-docs] win cmd (#1829)

DOMARS, 2019-10-23:
Thanks for reaching out to the Windows Driver Developer Documentation team. Unfortunately, we don’t have the resources to offer individual technical advice to the broad array of customers who use the driver documentation.
You may have success posting your question in the following forums, where other driver developers and Microsoft engineers can be found. The community there is quite responsive.
•	[Windows Hardware WDK and Driver Development](https://social.msdn.microsoft.com/Forums/windowsdesktop/en-US/home?forum=wdk)

closed 2019-10-23 by DOMARS