MicrosoftDocs/windows-driver-docs #1830: Inconsistent documentation of LSOv1 vs. LSOv2

I-gor-C, 2019-10-20:
Some items are mentioned as mandatory for both LSOv1 and LSOv2. Later there is a list of items that are necessary to be supported by LSOv2 in addition to LSOv1 items. 

Here is one:
> Any NET_BUFFER_LIST structure that contains the LSOv1 or LSOv2 OOB data also contains a single NET_BUFFER structure

But later:

> In addition to the previous LSO requirements, LSOV2-capable miniport drivers must also:
> * Limit the number of NET_BUFFER structures that are associated with each LSOV2 NET_BUFFER_LIST structure to one.

And also:
> Determine the total length of the packet from the length of the first NET_BUFFER structure in the NET_BUFFER_LIST structure.

It should be the same for LSOv1 if there is only one NET_BUFFER.

Another item:
> Use the byte offset in the TcpHeaderOffset member of NDIS_TCP_LARGE_SEND_OFFLOAD_NET_BUFFER_LIST_INFO to determine the location of the TCP header, starting from the first byte of the packet.

But [here](https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/content/ndis/ns-ndis-_ndis_tcp_large_send_offload_net_buffer_list_info) TcpHeaderOffset is explicitly defined for both LSOv1 and LSOv2.

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 39351457-2da4-e206-7f3e-2c20ff2b9c66
* Version Independent ID: ac5a694f-749f-1b0b-7161-b6c79e9e31de
* Content: [Offloading the Segmentation of Large TCP Packets - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/network/offloading-the-segmentation-of-large-tcp-packets)
* Content Source: [windows-driver-docs-pr/network/offloading-the-segmentation-of-large-tcp-packets.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/network/offloading-the-segmentation-of-large-tcp-packets.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @duncanmacmichael
* Microsoft Alias: **dumacmic**

duncanmacmichael, 2019-10-23:
Hey @I-gor-C, thanks for taking the time for this feedback. Upon further examination, though, I'm not sure where I see the inconsistencies. The first sample you pulled says that both LSOv1 and LSOv2 must have exactly one NB per NBL. The second sample says that LSOv2-capable miniport drivers must limit the NBL to having one NB. The third sample says that LSOv2-capable miniport drivers must be able to determine the total length of the packet from the first NB, which implies that there is exactly one NB (else they would have to traverse all the NBs to get the total length). The fourth sample says to use the **TcpHeaderOffset** member to determine the location of the TCP header...and the ref page to which you're linking has the exact same verbatim definition for that member for both LSOv1 and LSOv2.

So I'm sorry, but I don't see any contradictions here or any errors. Unless I'm missing something?

I-gor-C, 2019-10-23:
Hi @duncanmacmichael, I was referring to the fact that in some places these items are mentioned as additions of LSOv2 to the requirements from LSOv1, while in other places they are mentioned as mandatory for both LSOv1 and LSOv2. So it's not clear to me if they are mandatory for LSOv1 as well or not.

duncanmacmichael, 2019-10-24:
I see what you mean now, @I-gor-C. The list at the bottom of the topic is meant to define specific requirements for LSOv2 beyond what is already required for LSOv1. To that end, I've consulted with the NDIS team and re-read the doc so I can clarify these ambiguous points; those edits should be live soon. Let me know if this is still unclear. Thanks!

closed 2019-10-24 by duncanmacmichael