MicrosoftDocs/windows-driver-docs #1839: ZwClose call after ZwDeleteKey

drmkeeper, 2019-10-24:
"Once the driver has finished its manipulations, it must call ZwClose to close the handle—unless it has already called ZwDeleteKey to delete the key. (Once a key is deleted, all the open handles to it become invalid, so the driver must not close the handle in this case.)"
But also there is "After a call to ZwDeleteKey invalidates the key handles, you must call ZwClose to close the key handles." from 
https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/wdm/nf-wdm-zwdeletekey

So, which one is correct?

---
#### Сведения о документе

⚠ *Не вносите правки в этот раздел. Это необходимо для связывания страницы сайта docs.microsoft.com с вопросом на GitHub.*

* ID: c3f638cc-3575-9470-931e-0e14a3e432ab
* Version Independent ID: 494be0f1-eea1-1a8b-8451-46539facee90
* Content: [Using a Handle to a Registry-Key Object - Windows drivers](https://docs.microsoft.com/ru-ru/windows-hardware/drivers/kernel/using-a-handle-to-a-registry-key-object#feedback)
* Content Source: [windows-driver-docs-pr/kernel/using-a-handle-to-a-registry-key-object.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/kernel/using-a-handle-to-a-registry-key-object.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @tedhudek
* Microsoft Alias: **tedhudek**

tedhudek, 2020-05-06:
Hi @drmkeeper, thanks for the feedback and apologies for the delay. The first statement you cite was incorrect. I just pushed an update that will appear on the live site within a few minutes. See 53d8bd47b for details.

closed 2020-05-06 by tedhudek

tedhudek, 2020-05-06:
The new text reads:

Once the driver has finished its manipulations, it must call ZwClose to close the handle—even if it has already called ZwDeleteKey to delete the key. (Once a key is deleted, all the open handles to it become invalid, however the driver must still close the handle.)