MicrosoftDocs/windows-driver-docs #1849: Mismatch between FltKernel and this page

rodwiddowson, 2019-10-26:
In this case I rather suspect that the documentation is correct.

In the 17763 version  of fltkernel (I need a stable dev environment so I've not moved off VS17) we see the following
```
        PEPROCESS ProcessId;        //  Only meaningful for FastIo locking operations.
        BOOLEAN FailImmediately;    //  Only meaningful for FastIo locking operations.
        BOOLEAN ExclusiveLock;      //  Only meaningful for FastIo locking operations.
```

The comments do not appear to be meaningful.  Indeed it is  quite impossible for `FailImmediately` to be set to FALSE in a FASTIO environment and still respect the FASTIO contract.

Like I said at the top I'm pretty convinced that the documentation is correct and the file is incrrect, but you might want to check with @cgallred 

Or just close this as mistaken.

---
#### Document details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 26be1b7e-0f4b-8e18-5106-5b1d85c02c03
* Version Independent ID: b68b918d-7c3b-1611-171a-44f56012ddd6
* Content: [FLT_PARAMETERS for IRP_MJ_LOCK_CONTROL union - Windows drivers](https://docs.microsoft.com/en-gb/windows-hardware/drivers/ifs/flt-parameters-for-irp-mj-lock-control)
* Content Source: [windows-driver-docs-pr/ifs/flt-parameters-for-irp-mj-lock-control.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/ifs/flt-parameters-for-irp-mj-lock-control.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @lorihollasch
* Microsoft Alias: **loriwhip**

cgallred, 2019-11-05:
@rodwiddowson is correct.  Those parameters are valid for both fast I/O and IRP operations.  The header's comments are not correct.  The documentation is fine.

rodwiddowson, 2019-11-05:
I’ll close this then,  thanks @cgallred and @lorihollasch

closed 2019-11-05 by rodwiddowson