MicrosoftDocs/windows-driver-docs #1855: Command change & Cert not found

rikoudosenin, 2019-10-29:
The command needs to start with just certmgr instead of having exe attached to it as both cmd and powershell denies it.
Also I have tried this out but the certificate does not show up in the list at all, tried it on my primary machine and on vm

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 1cb064e9-8078-8709-56aa-c2cc06e0cee4
* Version Independent ID: 42a425fb-7528-771f-a312-6e85c97bc466
* Content: [Installing Test Certificates - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/install/installing-test-certificates#feedback)
* Content Source: [windows-driver-docs-pr/install/installing-test-certificates.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/install/installing-test-certificates.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @tedhudek
* Microsoft Alias: **tedhudek**

RAJU2529, 2019-11-06:
afternoon I will check on two machines

RAJU2529, 2019-11-06:
@rikoudosenin. After checking on my machine, i confirmed what you said is correct

tedhudek, 2020-07-17:
Removed the ".exe" - trying to get more info on why the cert might not show up

tedhudek, 2020-07-20:
Sorry for the delay on this one. Is it possible that you were looking at the user store and not the machine store? I just pushed a few updates to the doc page. The commit should show up above in the conversation for this issue in a few moments if you want to review the changes. Thanks for the feedback!

closed 2020-07-20 by tedhudek