MicrosoftDocs/windows-driver-docs #1859: Possibly incorrect Driver Verified bugcheck code documentation

KSUMatt, 2019-10-30:
I got the following bugcheck (with Driver Verifier enabled):
0xC4 bug check code

- Arg1: 2000
- Arg2: fffff801e7121c5d, The address in the driver's code where the error was detected.
- Arg3: 0.
- Arg4: 4d4d4c43

According to https://docs.microsoft.com/en-us/windows-hardware/drivers/debugger/bug-check-0xc4--driver-verifier-detected-violation it seems that this should be as follows:

- Parameter 1: 0x2000 - The Storport miniport driver passed a bad argument (a NULL pointer) to the StorPortInitialize routine.
- Parameter 2: 0xfffff801e7121c5d - The first argument passed to the StorPortInitialize routine. This parameter is a pointer to the driver object that the operating system passed to the miniport driver in the first argument of the miniport driver's DriverEntry routine.
- Parameter 3: 0x0 (NULL) - The second argument passed to the StorPortInitialize routine. This parameter is a pointer to context information that the operating system passed to the miniport driver in the second argument of the miniport driver's DriverEntry routine.

However, opening this dump file in WinDbg and running !analyze -v spits out:
Bug check c4 - A device driver attempting to corrupt the system has been caught.

- Arg1: 0000000000002000, Code Integrity Issue: The caller specified an executable pool type. (Expected: NonPagedPoolNx)
- Arg2: fffff801e7121c5d, The address in the driver's code where the error was detected.
- Arg3: 0000000000000000, Pool Type.
- Arg4: 000000004d4d4c43, Pool Tag (if provided).

This seems to follow the error code listed on this page: https://docs.microsoft.com/en-us/windows-hardware/drivers/devtest/code-integrity-checking

How do I know that this is in fact a "Code Integrity" issue and not a StorPortInitialize() routine issue? (Spoiler: The driver isn't a Storport miniport driver so I am confident Windbg is providing the correct information).

Has some of the Storport error codes incorrectly made it's way to the C4 bug check document? Should they be here instead: https://docs.microsoft.com/en-us/windows-hardware/drivers/debugger/bug-check-0xf1--scsi-verifier-detected-violation ?

Do these code integrity bug check codes need to be moved to the C4 page?

Thanks!

P.S. - Is this Code Integrity bugcheck something to be concerned with?

DOMARS, 2020-01-24:
@KSUMatt - I need to check with a few folks, but yes it looks like the error code list does need to be updated. And 0x2000 is a CI and not a Storport issue as you experienced. So thanks for this issue.

In general, most all DV violations are worth fixing, to ship a reliable and secure driver. In this case creating a driver that is HVCI compliant will allow the driver to run in more secure (Hypervisor-protected Code Integrity.) environments. You can read more about creating HVCI drivers in https://docs.microsoft.com/en-us/windows-hardware/drivers/driversecurity/use-device-guard-readiness-tool

DOMARS, 2020-01-29:
@KSUMatt - The topic has been updated to properly list 0x2000 as a CI issue, so closing this issue. I worked with the dev team to look at all of the codes and made a number of additions and corrections beyond the issue that you encountered. Thanks for seeing this error, and logging this issue.

closed 2020-01-29 by DOMARS