MicrosoftDocs/windows-driver-docs #1861: "downloads the package from WMIS"

PaulKlerkx, 2019-10-31:
I am troubleshooting this process.  When you refer to the information in my subject, could you include more detail on that - where exactly is WMIS - URL, ports?
event logs of all our devices have messages in the logs saying the devices can't access so I'm assuming we need a Proxy/firewall config change to allow it, but can't find that required information.  error message being - "A connection to the Windows Metadata and Internet Services (WMIS) could not be established."

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 9d5f648a-f925-7bca-7bad-b97fded54baf
* Version Independent ID: e6a51140-c16d-8096-7888-2536b23a3ebc
* Content: [Installing Device Metadata Packages from WMIS - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/install/installing-device-metadata-packages-from-wmis#feedback)
* Content Source: [windows-driver-docs-pr/install/installing-device-metadata-packages-from-wmis.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/install/installing-device-metadata-packages-from-wmis.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @tedhudek
* Microsoft Alias: **tedhudek**

RAJU2529, 2019-11-11:
@PaulKlerkx. As per my knowledge, WMIS online sever contains OEM certified device drivers for particular drivers

tedhudek, 2020-07-18:
Thanks for the feedback and apologies for the delay. It's possible that group policy might be set to prevent the WMIS acquisition. I'm afraid I don't have any expertise with it though, so if this is still an issue, I'd recommend posting in the following forums, where other driver developers and Microsoft engineers can be found. The community there is quite responsive.

* [Windows Hardware WDK and Driver Development](https://social.msdn.microsoft.com/Forums/windowsdesktop/en-US/home?forum=wdk)
* [OSR's ntdev List](http://www.osronline.com/showlists.cfm?list=ntdev)

Again, thanks for reaching out. Best of luck with this issue.

closed 2020-07-18 by tedhudek