MicrosoftDocs/windows-driver-docs #1862: Non-compliance with RFC 2863

I-gor-C, 2019-10-31:
Here is a quote for discards and errors definition:

> **ifInDiscards**
> The dropped-receive-buffer error count. This is the same value that OID_GEN_RCV_DISCARDS returns.
> 
> **ifInErrors**
> The receive error count. This count is the same value that OID_GEN_RCV_ERROR returns.

On the [OID_GEN_RCV_DISCARDS ](https://docs.microsoft.com/en-us/windows-hardware/drivers/network/oid-gen-rcv-discards)page the discard count is defined with little more details and with clear reference to RFC 2863:

> The count is the dropped-receive-buffer error count as described in RFC 2863.

It's not clear what "dropped-receive-buffer error count" means since there is no such counter in the RFC.

Looking at the NDIS_STATISTICS_INFO struct in ntddndis.h header we see the following:
```
    ULONG64                     ifInDiscards;           // OID_GEN_RCV_ERROR + OID_GEN_RCV_NO_BUFFER = OID_GEN_RCV_DISCARDS
    ULONG64                     ifInErrors;             // OID_GEN_RCV_ERROR

```
So the discards here are clearly defined as a sum of no-buffer and error counters.

Looking back at the [RFC](https://tools.ietf.org/html/rfc2863), it seems non-compliant:

> ifInDiscards OBJECT-TYPE
>     SYNTAX      Counter32
>     MAX-ACCESS  read-only
>     STATUS      current
>     DESCRIPTION
>             "The number of inbound packets which were chosen to be
>             discarded even though no errors had been detected to prevent
>             their being deliverable to a higher-layer protocol.  One
>             possible reason for discarding such a packet could be to
>             free up buffer space.
> 
> ifInErrors OBJECT-TYPE
>     SYNTAX      Counter32
>     MAX-ACCESS  read-only
>     STATUS      current
>     DESCRIPTION
>             "For packet-oriented interfaces, the number of inbound
>             packets that contained errors preventing them from being
>             deliverable to a higher-layer protocol.

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 583b20c5-b24a-47f2-0e64-95776ff02381
* Version Independent ID: cc068707-921d-82b6-11fb-f2e21d260195
* Content: [OID_GEN_STATISTICS - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/network/oid-gen-statistics)
* Content Source: [windows-driver-docs-pr/network/oid-gen-statistics.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/network/oid-gen-statistics.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @duncanmacmichael
* Microsoft Alias: **dumacmic**

duncanmacmichael, 2019-10-31:
Hmm, this does seem like a deeper implementation question, since the header says what it says and that doesn't appear to match the RFC. I'll have to defer to the NDIS team on this one. Let me email them to see what I can find out.

jtippet, 2019-11-01:
Thanks for bringing this up.  As you've pointed out, it's inconsistent.  Here's the current state of things:
* The netvmini sample driver does not sum the two error counters, but the older xframii sample does sum them
* NDIS's own NDIS5-to-NDIS6 OID translation layer sums the two error counters
* The code comment in ntddndis.h suggests they're summed
* The documentation is certainly ambiguous, but my impression is that it was meant to suggest you do not sum the counters
* I skimmed the OS source code for consumers of this data, and I didn't see anyone who was *subtracting* the error counters, so it's not common for consumers to assume they're summed.

That inconsistency dates back to the design of NDIS's adaptation of RFC 2863.  Unfortunately we're about 16 years too late to go back and fix the NDIS6 design.  You are just in time, though, to give us a chance to fix this for the upcoming NetAdapter statistics API.

In the meanwhile, here's my guidance.  Let's retroactively declare that NDIS_STATISTICS_INFO::ifInDiscards is defined to be exactly equal to RFC2863's ifInDiscards.  This should be the least surprise for applications, since they probably are more likely to be familiar with the RFC than with the NT kernel driver API.  So, first, pretend the code comment in ntddndis.h says this:

```
typedef struct _NDIS_STATISTICS_INFO
{
    ULONG64                     ifInDiscards;           // OID_GEN_RCV_NO_BUFFER
    ULONG64                     ifInErrors;             // OID_GEN_RCV_ERROR
} NDIS_STATISTICS_INFO;
```

Next, let's update the docs for OID_GEN_RCV_DISCARDS to say that it is defined to be `ifInDiscards` from the RFC.  That's now unambiguous wording.  Actually, if possible, I'd like to see *all* these OIDs updated to refer unambiguously to the RFC:

```
RFC                OID
---------------------------------------------------------------------------------
ifInOctets         OID_GEN_BYTES_RCV
ifInUcastPkts      OID_GEN_DIRECTED_FRAMES_RCV
ifInNUcastPkts     OID_GEN_MULTICAST_FRAMES_RCV + OID_GEN_BROADCAST_FRAMES_RCV
ifInDiscards       OID_GEN_RCV_DISCARDS
ifInErrors         OID_GEN_RCV_ERROR
ifInUnknownProtos  [not supported by the NIC driver, since the NIC driver doesn't demultiplex protocols]
ifOutOctets        OID_GEN_BYTES_XMIT
ifOutUcastPkts     OID_GEN_DIRECTED_FRAMES_XMIT
ifOutNUcastPkts    OID_GEN_MULTICAST_FRAMES_XMIT + OID_GEN_BROADCAST_FRAMES_XMIT
ifOutDiscards      OID_GEN_XMIT_DISCARDS
ifOutErrors        OID_GEN_XMIT_ERROR
```
In addition to the two "NUcast = MULTICAST + BROADCAST" cases above, we have a few more OIDs that provide more detail than the RFC's counters do.  But their definition is fairly clear, and can still be specified in terms of the RFC:
```
(OID_GEN_DIRECTED_BYTES_RCV + OID_GEN_MULTICAST_BYTES_RCV + OID_GEN_BROADCAST_BYTES_RCV) = ifInOctets
(OID_GEN_DIRECTED_BYTES_XMIT + OID_GEN_MULTICAST_BYTES_XMIT + OID_GEN_BROADCAST_BYTES_XMIT) = ifOutOctets
```

Finally, we will make sure this gets done correctly in NetAdapter.  (We will not go back and change the NDIS5-to-NDIS6 translation layer: there are very few NDIS5 drivers around anymore, and anyone who is using them has bigger problems to worry about than counting their dropped packets.)

Thanks again.

duncanmacmichael, 2019-11-01:
Thanks for the detailed response, @jtippet. I've updated all the OIDs you listed to refer explicitly to their counterpart in the RFC; those changes should be live shortly. I'll go ahead and close this issue, so thanks again for helping us find this @I-gor-C. You've helped improve the future implementation of a NetAdapterCx feature. :)

closed 2019-11-01 by duncanmacmichael