MicrosoftDocs/windows-driver-docs #1863: WiFiDirectLegacyAPDemo_v1.0 work

IgorTsukerman, 2019-10-31:
Try use WiFiDirectLegacyAPDemo_v1.0 (https://go.microsoft.com/fwlink/?LinkId=617905)
This demo work good only if wi fi client get IP address from access point using  DHCP. If use static IP addresses in AP and client , connection between becomes unstable. Connections status reported as Error in this demo.
Maybe the reason is known?

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 8f123dfa-22cf-e183-1bbf-5ae9f5c5f3f3
* Version Independent ID: fe306b23-cf52-6a5b-1b05-b4bbe48505c1
* Content: [Wi-Fi Direct - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/partnerapps/wi-fi-direct#feedback)
* Content Source: [windows-driver-docs-pr/partnerapps/wi-fi-direct.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/partnerapps/wi-fi-direct.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @EliotSeattle
* Microsoft Alias: **eliotgra**

stevewhims, 2019-11-01:
Hi, @IgorTsukerman. It’s currently expected that the client will be disconnected if it doesn't complete the DHCP exchange.

IgorTsukerman, 2019-11-02:
Hi,

IgorTsukerman, 2019-11-02:
Hi, @stevewhims. Thanks for your reply. Why disconnect expected? When I start this access point (mobile hot spot) from Windows menu and assign IP address manually, connection work normally.Unfortunately, Windows        requires working internet to start AP. I need static IP assignment for my task, because need connection with special device with static IP address.

stevewhims, 2019-11-03:
Hi, @IgorTsukerman, I'll pass this question along to the feature team, and we'll get back to you.

stevewhims, 2019-11-04:
@IgorTsukerman, I wonder whether you can confirm something for me:

The feature team at Microsoft sounds pretty sure that they're already working directly with you on this question on a different thread. That other thread is an email thread directly between folks at your company and the feature team. Also, the feature team believes that you filed a bug about the issue/question (259687). Unfortunately, I can't confirm from here whether that's accurate. I have *your* name in this thread, and I have the name of a company (and three employee names which don't match yours) on the other. I don't want to mention the other company name here. But if you feel comfortable letting me what company you represent, then I'll be able to see whether it's the same, or whether your query is actually independent of that. Or, if what I've said makes you realize that you and that other customer and one and the same, then please let me know. Thanks!

Meanwhile, the feature team say that the only info we can give is that the API wasn’t designed with your use case in mind, but rather it expects the client to complete the DHCP exchange.

Thanks!
Steve

IgorTsukerman, 2019-11-05:
Hi, @stevewhims. 
I represent SDB SI (Special Design Bureau for Seismic Instrumentation) from Saratov, Russia. You can see our site www.skbsp.ru. I know for sure that **nо one** (except me) of our company raised such question on different thread here.  

**> rather it expects the client to complete the DHCP exchange.**
However, when I start this access point (mobile hot spot) from Windows menu (Parameters window) and assign IP address manually, connection work normally and data exchange is not interrupted.
In example (WiFiDirectLegacyAPDemo_v1.0 (https://go.microsoft.com/fwlink/?LinkId=617905) and Windows menu the same network connection and corresponding AP appears with the same state. 

I use Microsoft Hosted Network Virtual Adapter in Windows 7 & 8 for these purpose, but in Windows 10 it deprecated. 

Our seismic devices require an access point to exchange data with application. We use rugged tablets with a soft AP for compactness in the field. This seismic devices not have DHCP client and use fixed subnet and IP address. Maybe there is some workaround?

Thanks,
Igor

stevewhims, 2019-11-05:
Thanks for this info, @IgorTsukerman, this is very helpful! I'll pass this along to the feature team, and we'll get back to you.

stevewhims, 2019-12-03:
UPDATE: I sent a reminder email, but we're still working internally on getting an answer here. Thanks for your patience! -Steve

arnaud-t, 2020-02-13:
Hi, 
I have exactly the same problem with one of our applications.
The wifi device configured with static IP disconnects every 30 seconds, which seems to be the DHCP timeout.
Do you have a workaround?

Regards, 
Arnaud

stevewhims, 2020-06-25:
I apologize that our internal email discussion has reached the point where we're unable at this time to provide a solution that we would recommend. Any techniques that were proposed during our long discussion were felt to come with too many caveats to recommend. I don't have permissions to close this issue, but I will step away from actively working it since a solution doesn't seem to be forthcoming.

It might be worth pursuing a resolution via some other channel. Here are all of the developer support resources, for completeness.

- [Microsoft Q&A](https://docs.microsoft.com/answers/topics/uwp.html?sort=active&filter=all) is the right place to lobby for a change, a feature, or to make a suggestion.
- For community assistance with development questions, there are our [forums](https://social.msdn.microsoft.com/Forums/newthread?category=windowsapps&forum=wpdevelop&prof=required), and [StackOverflow](https://stackoverflow.com/questions).
- The [Windows developer support](https://developer.microsoft.com/windows/support) site explains other support options.
- The [Visual Studio developer community](https://developercommunity.visualstudio.com) is available, and responsive. The former User Voice feature is now there on the **Suggest a feature** tab, if you want to lobby for a change, a feature, or to make a suggestion.
- For developer issues, or suspected product bugs, there's [Microsoft Support](https://support.microsoft.com/contactus); a support engineer will look at the repro and see whether the issue is a Microsoft product bug, or something else. If it's a bug with a Microsoft product then the ticket will be free of charge.

-Steve

closed 2020-06-25 by EliotSeattle