MicrosoftDocs/windows-driver-docs #1866: Mô DBH

Dinhhai94, 2019-11-01:
Mô dòng lệnh dữ liệu DBH

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 421ada36-ca88-ab33-7b6b-6ff86a2233f6
* Version Independent ID: e2005151-722a-5bb8-9820-ab349765c6db
* Content: [DBH Command-Line Options - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/debugger/dbh-command-line-options)
* Content Source: [windows-driver-docs-pr/debugger/dbh-command-line-options.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/debugger/dbh-command-line-options.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @DOMARS
* Microsoft Alias: **domars**

DOMARS, 2019-11-02:
@Dinhhai94  - I'm not able to understand this request. Can you provide additional information?

Bạn có thể cung cấp thêm thông tin?

DOMARS, 2019-11-23:
Xin chào @Dinhhai94 - I'm not able to determine what specific issue you are seeing with this topic, using translated text strings. The comments may indicate command line samples, but we have a topic that provides DBH command line samples - https://docs.microsoft.com/en-us/windows-hardware/drivers/debugger/additional-dbh-examples

As we have many users of these tools, the best option for general questions, is to post a specific question on the forums.

• Windows Hardware WDK and Driver Development

Good luck with your debugging.

closed 2019-11-23 by DOMARS